Serial-Studio/Serial-Studio #1: Two accelerometer widgets will have the same title

fusedFET, 2021-01-14:
Hello,
here is a little bug I found when creating two widgets with accelerometer.
Both widgets will have the same title.

```
{
    "t": "Sensor processing",
    "g": [
        {
            "t": "Accelerometer Unfiltered",
            "w": "accelerometer",
            "d": [
                {
                    "t": "X",
                    "v": "0",
                    "u": "m/s^2",
                    "g": false,
                    "w": "x"
                },
                {
                    "t": "Y",
                    "v": "0",
                    "u": "m/s^2",
                    "g": false,
                    "w": "y"
                },
                {
                    "t": "Z",
                    "v": "%1",
                    "u": "m/s^2",
                    "g": true,
                    "w": "z"
                }
            ]
        },
        {
            "t": "Accelerometer Filtered",
            "w": "accelerometer",
            "d": [
                {
                    "t": "X",
                    "v": "0",
                    "u": "m/s^2",
                    "g": false,
                    "w": "x"
                },
                {
                    "t": "Y",
                    "v": "0",
                    "u": "m/s^2",
                    "g": false,
                    "w": "y"
                },
                {
                    "t": "Z",
                    "v": "%2",
                    "u": "m/s^2",
                    "g": true,
                    "w": "z"
                }
            ]
        }
    ]
}
```
[image]

alex-spataru, 2021-01-14:
Hi, I just reproduced your issue, I will notify you as soon as I fix the problem. Thanks for the bug report!

alex-spataru, 2021-01-14:
I just uploaded a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.7) that fixes this issue. Can you please confirm that the new version works for you too?

fusedFET, 2021-01-15:
Thank you very much!
It seems to be fixed!

BTW, great work, there is so much potential in this tool. Love it 👍

closed 2021-01-15 by fusedFET