Serial-Studio/Serial-Studio #3: Display errors and quit unexpectedly

jiangjiangjun, 2021-01-15:
Hello,
First，Thank you for you work!
here is a little bug I found when useing.

1. the title num more than 6. llike this.
`
"t":"Sensor Readings",
         "d":[
            {
               "t":"Temperature",
               "v":"%5",
               "g":true,
               "u":"°C"
            },
            {
               "t":"Altitude",
               "v":"%6",
               "u":"m"
            },
            {
               "t":"Pressure",
               "v":"%7",
               "u":"KPa",
               "g":true
            },
            {
               "t":"External Temperature",
               "v":"%8",
               "g":true,
               "u":"°C"
            },
            {
               "t":"Humidity",
               "v":"%9",
               "g":true,
               "u":"°C"
            },
            {
               "t":"NO6",
               "v":"%7",
               "u":"KPa",
               "g":true
            },
            {
               "t":"NO7",
               "v":"%8",
               "g":true,
               "u":"°C"
            },
            {
               "t":"NO8",
               "v":"%9",
               "g":true,
               "u":"°C"
            }
         ]
`
[image]

2. serial-studio quit unexpectedly when inptu Parameter error。like this and there will a  Error messages when use other tools.
[image]

alex-spataru, 2021-01-15:
Hi, I'll take a look at both issues & notify you of any progress. Thanks for the bug report!

alex-spataru, 2021-01-16:
@jiangjiangjun I did not manage to reproduce your issues. However, I noticed that Serial Studio became irresponsive from time to time when receiving invalid data or corrupt text from the serial port (see screenshot for details).

[image]

The culprit for the irresponsiveness of the application was the way that the console text was handled when Serial Studio received incoming data from the serial port. I did some changes in the UI and the irresponsiveness issue was fixed (at least in macOS & Windows).

**In your case, Serial Studio crashes just after selecting the serial port? Or some time after interacting with the serial port device?** (This can help us narrow the probable causes of your crash).

On the other hand, I could move the scrollbar for a larger data group, even in a virtual machine with reduced resources:

[image]

I tried moving the scrollbar by dragging it and with the mouse wheel, and both options worked. I honestly don't know what could be causing your issue in your "Sensor Readings" group. Just to have exactly the same data displayed as you, can you **please share with me your full JSON file?**

Greetings!

alex-spataru, 2021-01-16:
Also, could you please post a screenshot of the other tools that you used & the error messages displayed by them?

alex-spataru, 2021-01-16:
@jiangjiangjun I managed to reproduce & fix your scrollbar issue. It only happens when the number of datasets in a group is equal to 7 (at least with my display resolution & font configuration). Diff of the offending code:

[image]

Basically, scrolling will be enabled only if the [scrollbar size](https://doc.qt.io/qt-5/qml-qtquick-controls2-scrollbar.html#size-prop) in relation to content height is smaller than 1.

jiangjiangjun, 2021-01-17:
hello,@alex-spataru 
I tried again ，The problem still exists.
1.this is my JSON map file form Communication modes and i add "NO6"、“NO7”、“NO8”。
[example.zip](https://github.com/Serial-Studio/Serial-Studio/files/5825520/example.zip)

[image]

2. I found that this problem is related to serial tools。Some are normal, some are abnormal，You can try different serial port tools。this is my.
[image]

3. Modifying the installation path does not work。

alex-spataru, 2021-01-17:
@jiangjiangjun I just uploaded a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.8) that includes the fixes for the scrollbar. I also added an error handler function for the serial port. In the case that a critical error occurs, the user is notified via a message box & the serial port is closed. Otherwise, the error is logged.

The new release also has a "log" button, which allows users to directly open the log file for to simplify bug reports. Also, I added a Chinese translation using online tools. 

[image]

If possible, please test the [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.8) so that we can confirm if it fixes your issues. 

Also, if you find errors in the translations, you are very welcome to modify [`zh.ts`](https://github.com/Serial-Studio/Serial-Studio/blob/master/assets/translations/zh.ts) with [Qt Linguist](https://doc.qt.io/qt-5/qtlinguist-index.html) & create a pull request.

jiangjiangjun, 2021-01-19:
@alex-spataru I test the 1.0.9。
1. the scrollbar is normal。
2. quit unexpectedly still exists.
this is log file.
[Serial Studio.zip](https://github.com/Serial-Studio/Serial-Studio/files/5832879/Serial.Studio.zip)
[image]
3. i will create a pull request for zh.ts.

jiangjiangjun, 2021-01-19:
@alex-spataru i have some trouble about access github。i will resolve Later。 this is zh.ts that fix version.and you can replace。
[zh.zip](https://github.com/Serial-Studio/Serial-Studio/files/5833349/zh.zip)

alex-spataru, 2021-01-19:
Thanks for the logs, there are several serial port errors reported in the log file, I will check them one by one to see what can be done about it (or how can I handle the errors before the application crashes).

alex-spataru, 2021-01-19:
**Update:** I managed to crash Serial Studio while selecting a serial port 😄. Now that I know how to reproduce the issue, I'll get to the debugger and come with a possible fix.

alex-spataru, 2021-01-20:
@jiangjiangjun I uploaded a new release that handles serial port errors in such a way that the crash is avoided (at least in all my test scenarios). Can you please (again) check if the new release fixes your issue?

jiangjiangjun, 2021-01-20:
@alex-spataru  I test the 1.0.10。
the good news! the quit unexpectedly is fix。
[image]

the other problem is the COM port list not refresh 。COM12 was removed 。

Thank you again for your work。

jiangjiangjun, 2021-01-26:
test v.1.0.11,the COM port list not refresh is fix。the issue can close.

alex-spataru, 2021-01-26:
Perfect, thanks a lot of the bug reports and the contributions! Closing this issue now.

closed 2021-01-26 by alex-spataru