Serial-Studio/Serial-Studio #5: [Feature Request] FFT/DFT signal plotting

fusedFET, 2021-01-18:
Hello, 
I have been using this tool to visualize in real time the effect of DSP on the device.
(Which was very usefull by the way!)

** Problem **
If I have data which needs to be shown that is not in the time domain.
For example: FFT/DFT which processed on the device or 1D cameras.

** Solution **
2 Widgets where the Y axis is not time based.
- A classic line-chart, based on a table.
- a heatmap/waterfall diagramm to show for example 1D grayscale diagrams in time or an FFT/DFT in time.

The device would just send the row/column count followed by values set in one transmission.

**Describe alternatives you've considered**
Exporting signal data to Scilab/Octave, which is tedious but is also usefull.

alex-spataru, 2021-01-18:
Hi @fusedFET, don't worry - it's part of the issue template that I created yesterday.

I'll take a look at how to generate the FFT diagrams and how should we need to modify the `*.json` file in order to achieve that. I'll keep you notified through this thread.

Greetings!

alex-spataru, 2021-04-20:
Closing this issue, discussion will be continued in #49.

closed 2021-04-20 by alex-spataru