Serial-Studio/Serial-Studio #11: compile errors, missing files

Andromem, 2021-01-29:
Compile error!
   "No such file or directory"
These files are missing.

please upload these files:

<qtcsv/stringdata.h>
<qtcsv/reader.h>
"Logger.h"
"ConsoleAppender.h"

alex-spataru, 2021-01-30:
Hi, this repository makes use of [submodules](https://git-scm.com/book/en/v2/Git-Tools-Submodules) for all the third-party libraries that it uses. Unfortunately, downloading the ZIP file from GitHub won't include these third-party repositories.

As stated in the [README](https://github.com/Serial-Studio/Serial-Studio#build-instructions), you need to clone the repository using the following commands:

```
git clone https://github.com/Serial-Studio/Serial-Studio
cd Serial-Studio
git submodule init
git submodule update
```

You can also use `--recursive` option for `git clone`:

`git clone --recursive https://github.com/Serial-Studio/Serial-Studio`

Please try executing those commands before building the application. If you need any help, please let me know.

alex-spataru, 2021-01-31:
@Andromem You're welcome! Shall I close the issue?

closed 2021-02-01 by alex-spataru