Serial-Studio/Serial-Studio #12: Support arbitrary baud rates

MCUdude, 2021-01-31:
Hi!

It would be great if Serial Studio supported user-defined baud rates. In many cases, an application uses a different baud rate that's currently available. For instance, I often use 250000 and 500000 baud on 16 MHz microcontrollers. Sometimes even 1M. The midi baud rate is 31250 baud, and other common ones are 460800 and 921600 baud.

marcobrianza, 2021-02-01:
hi, same here. 250000 is my working speed on AVR

alex-spataru, 2021-02-01:
Hi, I experimented with custom baud rates on other Qt-based projects, I just need to update the UI code. I'll post progress updates in this thread. Thanks for the idea!

alex-spataru, 2021-02-01:
I just made a commit that allows users to select a custom baud rate value (see screenshot below):

[image]

Please let me know if you want to change anything from this screenshot. This feature will be available in the next release (1.0.13).

marcobrianza, 2021-02-01:
looks good to me, thankyou

MCUdude, 2021-02-01:
@alex-spataru thank you for your quick reply and your willingness to listen what other "dumb" users have to say.
I do maintain some kinda large Github projects myself, and I know you can't implement everybody's wishes; that would make the UI terrible 😉 

I don't know anything about Qt and what limitations it has, but would it be possible to have a dropdown menu like the latest release has, but where you can type your user defied baud rate _in_ the same field? That would eliminate the extra field in your screenshot and the check button.

Also, I'm not sure if the UI adds the comma to the custom baud rate, or you typed it in, but its not common to have a comma there. At least I've never seen it in any terminal program yet.

alex-spataru, 2021-02-01:
I think that editable comboboxes are supported by Qt. I'll take a look at that option, however, I think that the most intuitive thing for new users would be to have two separate controls. The UI control adds a comma automatically, but it sends a integer to the C++ serial manager class. I'll change the display format that to avoid confusion.

alex-spataru, 2021-02-02:
It turns out that the editable combobox was a good idea, it also allows us to save custom baud rates between runs. Here's a screenshot of how it looks now:

[image]

MCUdude, 2021-02-02:
Looks very good! Is the user defined baud rate "remembered" when you close the application and re-open it?

I'm not sure how the complete list looks like at the moment, but It would totally make sense to add 250000, 460800, 500000, 921600 and 1000000 baud to the list of "standard" baudrates, since they really are commonly used, espcially on microcontrollers.

Then it would make sense to let the user type in "weird" baud rates such as 31250 (midi), 56000 and 74880 (ESP8266 bootloader baud rate).

alex-spataru, 2021-02-02:
> Looks very good! Is the user defined baud rate "remembered" when you close the application and re-open it?

Thank you! The user-defined baud rate is saved when you close the application and re-open it, the baud rate list is automatically sorted to avoid cluttering too much the UI after working with a lot of custom baud rates.

> I'm not sure how the complete list looks like at the moment, but It would totally make sense to add 250000, 460800, 500000, 921600 and 1000000 baud to the list of "standard" baudrates, since they really are commonly used, espcially on microcontrollers.

I just added those baud rates to the default baud rate list with the last commit. Thanks a lot of the suggestion!

MCUdude, 2021-02-02:
Thanks for listening! The reason why I'm suggesting all this is because I hope Serial Studio can replace versatile terminals such as Realterm (Windows) and CoolTerm (Mac).

alex-spataru, 2021-02-02:
You're welcome! Bugs and improvements play an important role in the development cycle of any software. Please let me know if the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.13) fixes this issue!

MCUdude, 2021-02-02:
Just downloaded 1.0.13, and the custom baud rate implementation turned out real good! Thanks for adding it!

closed 2021-02-02 by MCUdude