Serial-Studio/Serial-Studio #14: Windows installer force restarts PC!

mrx23dot, 2021-02-01:
Windows installer of Serial Studio 1.0.12 release instantly force restarted my windows without asking at the end.
Lost a few hours of works. wtf

alex-spataru, 2021-02-01:
Hi, I have tested the installers on several Windows installations, and I did not experience this issue. Could you please provide me with more information about your computer/Windows installation?

Also, could you please tell me if the restart occurred after you clicked on the "Finish" button (see screenshot below):

[image]

Or did it occur during the installation of Visual Studio redistributable? (see screenshot):

[image]

The only feasible explanation that I can come up with is that for some reason the Visual Studio Redistributable installer forced the computer to restart. I'll add the `/norestart` option to the installer for future releases just to be sure.

mrx23dot, 2021-02-01:
For a brief moment I think I saw the first screen with the checkbox, but didnt have time to read and get prepared.
Thanks for the change.

alex-spataru, 2021-02-02:
This issue should be fixed with the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.13). Indeed the `/norestart` option is necessary to avoid these kind of problems. Please let me know if we should close this issue.

closed 2021-02-02 by mrx23dot