Serial-Studio/Serial-Studio #15: Missing Qt dependencies in the Build instructions

Foalyy, 2021-02-01:
**Describe the bug**
Just an improvement to the README : there are two Qt dependencies that don't appear to be explicitly mentioned in the build instructions, QtCharts and QtGraphicalEffects. I don't know about other systems, but they are not provided by default on my ArchLinux system, I had to look into the build log and install qt5-charts and qt5-graphicaleffects from the repositories. On Windows, this may require checking the correct boxes during Qt's install? I'm not sure about this.

**To Reproduce**
Steps to reproduce the behavior:
Follow the Build instructions on a new Qt install.

**Expected behavior**
Dependencies explicitly stated in the README.

alex-spataru, 2021-02-01:
Whoops I totally forgot to mention that in the readme. Thanks for the report!

DanielPrjct, 2021-02-01:
I'm using Arch Linux too.
I needed to install **qt5-quickcontrols** to launch Serial-Studio properly.
Otherwise, it complains:
 [QQmlApplicationEngine failed to load component                           [W]
qrc:/qml/main.qml:24:1: module "QtQuick.Dialogs" is not installed        [W]
and Serial-Studio graphical interface aborts...

alex-spataru, 2021-02-01:
I just updated the [README](https://github.com/Serial-Studio/Serial-Studio#build-instructions) files. Could you please corroborate that the new build instructions "fix" this issue?

DanielPrjct, 2021-02-02:
Dear Alex Spataru:

I've checked Serial Studio dependencies in Arch Linux:

[daniel@hardwork ~]$ objdump -p /home/daniel/Serial-Studio/serial-studio | grep NEEDED
  NEEDED               libQt5Svg.so.5
  NEEDED               libQt5Charts.so.5
  NEEDED               libQt5Widgets.so.5
  NEEDED               libQt5QuickControls2.so.5
  NEEDED               libQt5Quick.so.5
  NEEDED               libQt5Gui.so.5
  NEEDED               libQt5Xml.so.5
  NEEDED               libQt5Sql.so.5
  NEEDED               libQt5QmlModels.so.5
  NEEDED               libQt5Qml.so.5
  NEEDED               libQt5Network.so.5
  NEEDED               libQt5SerialPort.so.5
  NEEDED               libQt5Core.so.5
  NEEDED               libGL.so.1
  NEEDED               libpthread.so.0
  NEEDED               libstdc++.so.6
  NEEDED               libm.so.6
  NEEDED               libgcc_s.so.1
  NEEDED               libc.so.6

The packages that provides these dependencies in Arch Linux (and other Arch derivative distros like Manjaro) are:

    libc
    gcc-libs
    libglvnd
    qt5-base
    qt5-serialport
    qt5-declarative
    qt5-quickcontrols2
    qt5-charts
    qt5-svg

---

Below, the complete output of  lddtree <file>, and pacman -Qo <file> for serial-studio deps:

[daniel@hardwork ~]$ lddtree /home/daniel/Serial-Studio/serial-studio
/home/daniel/Serial-Studio/serial-studio (interpreter => /lib64/ld-linux-x86-64.so.2)
    libQt5Svg.so.5 => /usr/lib/libQt5Svg.so.5
        libz.so.1 => /usr/lib/libz.so.1
    libQt5Charts.so.5 => /usr/lib/libQt5Charts.so.5
    libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5
    libQt5QuickControls2.so.5 => /usr/lib/libQt5QuickControls2.so.5
        libQt5QuickTemplates2.so.5 => /usr/lib/libQt5QuickTemplates2.so.5
    libQt5Quick.so.5 => /usr/lib/libQt5Quick.so.5
    libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5
        libpng16.so.16 => /usr/lib/libpng16.so.16
        libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0
            libfreetype.so.6 => /usr/lib/libfreetype.so.6
            libharfbuzz.so.0 => !!! circular loop !!!
                libbz2.so.1.0 => /usr/lib/libbz2.so.1.0
            libgraphite2.so.3 => /usr/lib/libgraphite2.so.3
            libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0
                libpcre.so.1 => /usr/lib/libpcre.so.1
        libmd4c.so.0 => /usr/lib/libmd4c.so.0
    libQt5Xml.so.5 => /usr/lib/libQt5Xml.so.5
    libQt5Sql.so.5 => /usr/lib/libQt5Sql.so.5
    libQt5QmlModels.so.5 => /usr/lib/libQt5QmlModels.so.5
    libQt5Qml.so.5 => /usr/lib/libQt5Qml.so.5
    libQt5Network.so.5 => /usr/lib/libQt5Network.so.5
        libdl.so.2 => /usr/lib/libdl.so.2
        libgssapi_krb5.so.2 => /usr/lib/libgssapi_krb5.so.2
            libkrb5.so.3 => /usr/lib/libkrb5.so.3
            libk5crypto.so.3 => /usr/lib/libk5crypto.so.3
            libcom_err.so.2 => /usr/lib/libcom_err.so.2
[... 51 lines cut ...]
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libQt5QmlModels.so.5
/usr/lib/libQt5QmlModels.so.5 is owned by qt5-declarative 5.15.2-1
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libQt5Qml.so.5
/usr/lib/libQt5Qml.so.5 is owned by qt5-declarative 5.15.2-1
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libQt5Network.so.5
/usr/lib/libQt5Network.so.5 is owned by qt5-base 5.15.2-3
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libQt5SerialPort.so.5
/usr/lib/libQt5SerialPort.so.5 is owned by qt5-serialport 5.15.2-1
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libQt5Core.so.5
/usr/lib/libQt5Core.so.5 is owned by qt5-base 5.15.2-3
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libGL.so.1
/usr/lib/libGL.so.1 is owned by libglvnd 1.3.2-1
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libpthread.so.0
/usr/lib/libpthread.so.0 is owned by glibc 2.32-5
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libstdc++.so.6
/usr/lib/libstdc++.so.6 is owned by gcc-libs 10.2.0-4
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libm.so.6
/usr/lib/libm.so.6 is owned by glibc 2.32-5
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libgcc_s.so.1
/usr/lib/libgcc_s.so.1 is owned by gcc-libs 10.2.0-4
[daniel@hardwork ~]$ pacman -Qo /usr/lib/libc.so.6
/usr/lib/libc.so.6 is owned by glibc 2.32-5

Best regards,
Daniel

alex-spataru, 2021-02-02:
@DanielPrjct Thanks a lot for the information, I think that the correct approach would be to create a wiki entry for compilation instructions and dependencies for Linux distros, and provide a link in the README files. The package names probably change between Ubuntu/Fedora and Arch-based distros (and thats the reason I chose to distribute as an AppImage, it takes a lot of work and time to properly support & test each distro).

Nevertheless, I think that specifying the necessary packages/dependencies and listing the compilation commands (`qmake`, `make` & `sudo make install`) would be a good idea for users that don't like the AppImage format.

DanielPrjct, 2021-02-03:
Estimado Alex:

Te contesto en español porque también es mi idioma nativo.
Nací y vivo en Argentina, aunque tuve la oportunidad de conocer un poco de
México en un viaje turístico hace unos diez años atrás.

Respecto a Serial-Studio: no me había dado cuenta de que en la seccion
Releases estaba disponible la versión 1.0.13 en formato Appimage. Mañana
sin falta la descargaré y probaré.

¡Muchas gracias por tu trabajo!

Coincido totalmente con tu enfoque de distribuir el software que
desarrollaste en ese formato, para evitar las complicaciones que genera la
enorme diversidad y dispersión entre los sistemas de paquetes de las
distintas distros de Linux, sumadas a los inexplicables cambios que algunas
imponen en los nombres de paquetes que proveen el mismo
programa/biblioteca/librería.

Después de varios años usando Debian, Ubuntu, Mint y otras derivadas de la
primera, recalé en Arch Linux por su simpleza, su actualización permanente
(rolling release) y también por ser más "upstream compliant" que otras
distros.

A modo de ejemplo: en mi sistema he instalado FreeCAD utilizando su
.appimage, me ha ahorrado horas y horas de compilación del programa y de
sus (pesadas) dependencias, ya que no están disponibles en los repos
binarios oficiales de Arch...

Pero otras aplicaciones que me resultan imprescindibles las he tenido que
compilar utilizando los PKGBUILD que distintos usuarios han publicado en
Arch Users Repository (BTW: yo mantengo USBPICPROG y TinyBootloaderLin en
el AUR).

Te envío un gran saludo desde Mar del Plata,
Daniel

umarcor, 2021-02-04:
Hola Daniel!

FYI, in #9 you will find a PKGBUILD recipe for MSYS2. It's not exactly the same as the one you need for Arch, but it's pretty similar. MSYS2's package manager is `pacman`, a port of Arch's. Hence, MSYS2 is also a rolling release.

alex-spataru, 2021-02-04:
@DanielPrjct ¡Muchas gracias por el agradecimiento! Es un gusto saber que un proyecto personal/escolar sea capaz de ayudar a personas en otros lados del mundo. Personalmente todavía no he visitado Argentina, pero es algo que me gustaría hacer en esta vida.

Espero que la versión en formato AppImage le haya funcionado de manera satisfactoria. En todo caso, voy a estar trabajando en documentar el proceso de compilación e instalación en diferentes distros en la wiki del proyecto (probablemente el fin de semana).

¡Saludos desde México!

closed 2021-09-23 by alex-spataru