Serial-Studio/Serial-Studio #18: [Feature request] Add connect/disconnect button

MCUdude, 2021-02-01:
Hi!

The user interface is already quite nice, and I like the idea that you automatically connect to the serial port when you select it. However, it would be nice if there was a connect/disconnect button in the toolbar. The reason this would be practical is when you need to re-upload code to your microcontroller. At the moment you'll have to select another serial port and re-select it to reconnect. A simple connect/disconnect button (green/red button perhaps?) would make this process even more elegant.

Another idea: why not have read/write mode selected by default? In most cases it's not harmful to write to the connected device.

alex-spataru, 2021-02-01:
Hi, I am currently using this approach for other (proprietary) projects. Here are some screenshots of the button, which is located at the toolbar: 

[image]

[image]

Please let me know if you like that design, or if have another idea. I'll change the default open mode to RW to avoid confusion among new users of the project.

MCUdude, 2021-02-01:
That looks very good!

> I'll change the default open mode to RW to avoid confusion among new users of the project.

Excellent! I'm sure this would suit most users.

alex-spataru, 2021-02-02:
I did some UI revamps and added the conenct/disconnect button. Here are the screenshots:

[image]

[image]

Having this button is indeed more intuitive than changing the serial devices combobox. And it can be accessed even if the setup pane is hidden.

alex-spataru, 2021-02-02:
Hi, this feature is now available with the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.13). Please let me know if we should close this issue.

MCUdude, 2021-02-02:
This turned out REALLY GOOD! Issue can definitly be closed.

closed 2021-02-02 by MCUdude