Serial-Studio/Serial-Studio #19: Map provider issue on Windows build and log error

helgrind, 2021-02-02:
I'm trying to show GPS coordinates on the latest Windows build and am seeing the following in my log. The map widget shows a white box with an empty drop-down. I'm not sure if the argument missing is due to my json or not? Thanks in advance for this great tool!

`
[23:49:15.757] QString::arg: Argument missing: {
   "t":"PIPICOGPS",
   "g":[
      {
         "t":"GPS",
         "w":"map",
         "d":[
            {
               "t":"Longitude",
               "v":"51.6151",
               "u":"°N",
               "w":"lon"
            },
            {
               "t":"Latitude",
               "v":"1.227836",
               "u":"°W",
               "w":"lat"
            }
         ]
      }
   ]
}, 0.0 [W] []
[23:49:29.388] Loading JSON file, silent flag set to false                              [I] [JsonGenerator::loadJsonMap]
[23:49:29.389] JSON map loaded successfully                                             [I] [JsonGenerator::loadJsonMap]
[23:49:29.440] 1 ,  "The geoservices provider is not supported."                        [D] []
`

alex-spataru, 2021-02-02:
Hi, you are not missing anything on your JSON. I just reproduced your issue on Windows, I am currently testing a possible fix. A quick side-note: the map widget expects the latitude to be negative if it "points" west. Thus, the appropriate value should be. `-1.227836`.

helgrind, 2021-02-02:
Ah, thanks. Yes that would be correct for decimal degree format. Silly me!

alex-spataru, 2021-02-02:
I think I got the issue. Serial Studio uses `mapboxgl` as its default map provider. Unfortunately, this is not supported for Visual Studio builds ([QTBUG-78715](https://bugreports.qt.io/browse/QTBUG-78715)). I'll use [Open Street Map](https://www.openstreetmap.org/) as the default map provider on Windows.

alex-spataru, 2021-02-02:
Hi, the issue should be fixed with the latest release. I also added support for tilting the map:

[image]

Please let me know if the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.13) also works on your computer, so that I can close this issue.

helgrind, 2021-02-02:
Thanks, looks good but the coordinates seem to place me off the coast of SA, when I'm in the UK.
Checked against Google maps and they were good coords: 51.615N, -1.228W

[image]

helgrind, 2021-02-02:
Oh dear, that was my bad. Reversed of course. Thanks for sorting it, all done.

closed 2021-02-02 by helgrind

alex-spataru, 2021-02-02:
Don't worry, I just was going to say the same thing. Thanks for the bug report!