Serial-Studio/Serial-Studio #20: [Feature request] Use up and down arrow keys to select previously sent messages

MCUdude, 2021-02-02:
This one is super useful and is present in the Arduino IDE serial monitor.

When the send-text field is highlighted, it would be awesone if one could press the up and down keys to cycle between older messages that have been sent previously, just like you can on "standard" terminal application like CMD on windows and Terminal on Mac.

It would be awesome if this could apply to the HEX mode as well!

Speaking of HEX mode, is this a bug or a feature? I'm able to place the cursor anywhere I want, even between every nibble. Shouldn't this be left oriented, just like in ASCII mode? That's atleast what Realterm does.

Screenshot for reference:

[image]

alex-spataru, 2021-02-03:
The history feature should not be too hard to implement. Regarding HEX mode, I just confirmed that your issue happens on my computer too, I'll fix it later this week.

alex-spataru, 2021-02-05:
**Update:** as of the latest commit this feature was also implemented. I'll notify you when I upload the next release so that this feature can be tested.

MCUdude, 2021-02-05:
Awesome! Were you able to get rid of the bug where the cursor could be set anywhere in hex mode as well?

alex-spataru, 2021-02-05:
> Awesome! Were you able to get rid of the bug where the cursor could be set anywhere in hex mode as well?

As of the latest commit, yes. I changed the method of validating user input when using hex mode.

alex-spataru, 2021-02-17:
I'll close this issue because this feature has been implemented. If you have any suggestions, or would like to change something, please post an update in the UI enhancements thread (#21). Greetings!

closed 2021-02-17 by alex-spataru