Serial-Studio/Serial-Studio #21: [Feature request] Console suggestions

MCUdude, 2021-02-02:
Sorry for creating so many issues, but I see a lot of potentioal in this project, and in my opinion the console itself it a very good general purpose serial monitor if it had a few more features. Here are some ideas you can take into consideration:

* Everything I type and send gets echoed back on screen without any indication that this is the echo, not a reply from the attached serial device. The echo should IMO be optional, and default off, just like other serial monitors.
* There should also be a way to automatically add [LF], [CR] or [CRLF] to the message sent from Serial Studio. Ideally from a dropdown menu just like in Arduino IDE, or perhaps a checkbox or similar.
* It would be great if one could change the baudrate without having to disconnect, change and reconnect again. Arduino IDE let's you do this on the fly, and it's very practical if you accidentally selected the wrong baud rate. It's the small things that makes the user experience 🙂 
* It would also be really neat if there was a way to disable autoscrolling in the console. Sometimes you're logging somthing, and you'd like to stop and take a closer look without disconnecting from your serial device.

Here's how [CR]/[LF] is done in Coolterm (which is not a very intuitive terminal program):

[image]

Here's the timestamp setting in the Arduino IDE serial monitor:

[image]

alex-spataru, 2021-02-03:
Thanks a lot for the suggestions, I'll take a look at them and implement them over the course of the week.

jakkra, 2021-02-03:
Hijacking this issue with another relevant feature request regarding using this as a serial console and that is the be able to select text in the serial window and to be able to copy it. Something I use almost daily. I see this have potential to be my daily driver for serial stuff. Also +1 on optional echo when something is sent in the console.

MCUdude, 2021-02-03:
> Hijacking this issue with another relevant feature request regarding using this as a serial console and that is the be able to select text in the serial window and to be able to copy it. 

Whoa, I didn't realize it currently isn't possible to mark and copy text in the console window. This is a must have!

BTW did this project go viral or something? I mean, look at the Github stars! They increase every time I update the website 😮

alex-spataru, 2021-02-04:
> Hijacking this issue with another relevant feature request regarding using this as a serial console and that is the be able to select text in the serial window and to be able to copy it.

> Whoa, I didn't realize it currently isn't possible to mark and copy text in the console window. This is a must have!

I did not catch that one either, for some reason I thought that the Qt controls already handled that. Thanks for the heads-up! 

I am currently working on organizing the codebase to make the project more maintainable in the future (and being able to support interfacing with devices through a network socket [#17]), so it might take a bit until I push my changes to the repo.

However, I am already working on this issue by implementing most of the console features in C++ (instead of relying on QML for the console logic).

alex-spataru, 2021-02-04:
I just implemented most of the features requested (I still have to check the part of selecting text & copying it). Here's a screenshot of how it looks right now:

[image]

Please let me know what you think about it before beginning to work on other issues.

MCUdude, 2021-02-04:
I haven't tried it myself, but it looks very good. Is it possible to change baudrate on the fly too?

I'm also looking forward to see what #20 will bring 🙂

alex-spataru, 2021-02-05:
@MCUdude Yes, I added support for changing the baud rate on the fly. I'll notify you when I release the new version!

alex-spataru, 2021-02-08:
Hi, I just uploaded the [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.14). Please let me know how the console features work for you!

jakkra, 2021-02-08:
This is kinda picky, but I'm not able to select individual word in a line, I'm only able to copy the full line. Awesome with save feature by the way!

alex-spataru, 2021-02-08:
> This is kinda picky, but I'm not able to select individual word in a line, I'm only able to copy the full line.

I get you, unfortunately the text editor components offered by QtQuick use way to many resources for logging in realtime (the application would begin to slow down at about ~100 lines), so I practically had to implement a home-made text editor. 

Individual word selection will be eventually implemented, but I wanted to upload the release to get feedback on the current changes. I'll keep the issue open, because there are still a lot of things to improve/fix regarding the console widget.

> Awesome with save feature by the way!

Thanks! Just take into consideration that the maximum scroll-back is currently 5000 lines. In the case that the current console output exceeds 5000 lines, Serial Studio will begin removing the first lines of the buffer in order to keep resource usage in check.

If you expect to work with larger outputs, just be sure to save the console output from time to time in a different file (I am currently working on opening the output file in append mode to be able to save all output in the same file).

MCUdude, 2021-02-08:
Thanks @alex-spataru, 1.0.14 is a huge improvement!
I have few thoughts though:

* I miss the byte-oriented HEX view when I type. It's much easier to read what's in the text field when every byte has a space between them, just like in the previous version, 1.0.13. Is is possible to get this back?
* It would also be great if the text field stayed highligheted when you change the settings below the text field. This would mean that one could quickly change settings without having to click in the text field to be able to continue typing.
* It would also be fantastic if it could select the previously used serial port if available. If not available, show "Select port", just like it is right now.
* It would be nice if the shortcuts shown in the console window was present in the toolbar, here:
  - [image]

> Just take into consideration that the maximum scroll-back is currently 5000 lines. In the case that the current console output exceeds 5000 lines, Serial Studio will begin removing the first lines of the buffer in order to keep resource usage in check.

This is definitly a hard limit. Logging in a serial monitor is quite common, so this is definitly bad news. Another thing, I created a simple program on an Arduino that prints 6000 lines before it stops. After it's done it is very difficult to scroll in the console window. I can sometimes scroll up, but I can't scroll down without clicking in the scrollbar on the right side.

alex-spataru, 2021-02-10:
Update, I just found a way to insert classical widgets into the QML interface, which drastically increases the performance of the application when dealing with large serial input. As of the latest commit, all of the console buffer can be exported as a text file (regardless of the line count). Here's a screenshot of how it looks the new text editor:

[image]

> I miss the byte-oriented HEX view when I type. It's much easier to read what's in the text field when every byte has a space between them, just like in the previous version, 1.0.13. Is is possible to get this back?

Yes, I'll add a handler to the text input field to automatically add a space every two characters.

> It would also be great if the text field stayed highligheted when you change the settings below the text field. This would mean that one could quickly change settings without having to click in the text field to be able to continue typing.

I think this can be done by changing the focus properties in the QML interface, but I am not very sure. I'll take a look at that & keep you updated.

alex-spataru, 2021-02-14:
Hi, I just released a [new version](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.16) that implements many improvements in the console widget. I would appreciate a lot your feedback about the new release, since it fixes many of the problems mentioned in this thread.

jakkra, 2021-02-16:
Looks good, I'll use it for the day and report back if I find some issues, thanks! 
So far I only see one very minor thing and that is the autoscroll is offset about 3 lines. Not always 3 lines, seems to be depending on window resize as it seems to change how many lines it shows.

closed 2021-09-23 by alex-spataru