Serial-Studio/Serial-Studio #22: CSV Player Filename and Timestamps are UTC not Local Time

TDStrasser, 2021-02-03:
Congratulations on a very functional app which solves numerous problems for real-time data visualisation for STEM classes and projects.  The CSV Player feature is great except the timestamps applied are in UTC not in the computer localtime, the former being confusing and the latter more useful.
I have tried this both in Windows and Macintosh computers and the bug is the same.  I am in Australia at UTC+10 and the timeshift here is very noticeable.
Otherwise, great job!
Regards,
Tony

alex-spataru, 2021-02-03:
Hi, I just made a commit that fixes your issue, values on `RX Date/Time` are now generated using the computer's local time. This feature will be available in the next release of the application. Thanks for the report!

TDStrasser, 2021-02-03:
Thank you for the quick response!

alex-spataru, 2021-02-08:
Hi, I just uploaded the [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.14). Please let me know if it fixes your problems (so that I can close this issue).

TDStrasser, 2021-02-08:
Thanks Alex. The timestamps and records are now in local time. However now on CSV playback the display is no longer per the loaded JSON file.  The case in point is a 3-axis accelerometer with X, Y and Z axis acceleration variables in a group and each with its own graph.  This format is not followed when the recorded CSV file is loaded and played.

alex-spataru, 2021-02-08:
I just managed to reproduce your issue. I'll take a look at this problem later today. Thanks for the bug report!

alex-spataru, 2021-02-08:
I just uploaded a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.15) that fixes your issue. Please let me know if it works for you too.

TDStrasser, 2021-02-08:
Thanks for your quick response Alex.  That fix seems to have worked with the replay now using the JSON definition file.  
There is still a small bug. The CSV player control pop-up window "play" button has the incorrect icon on startup (pause instead of play). The file is not actually playing until that button is clicked. If then click to pause the correct "play" icon then appears.
Also this pop-up window overlays all other windows and does not disappear if another app is brought into focus.
[image]

alex-spataru, 2021-02-09:
> The CSV player control pop-up window "play" button has the incorrect icon on startup (pause instead of play). The file is not actually playing until that button is clicked. If then click to pause the correct "play" icon then appears.

Again, thanks for the report. For the moment I cannot reproduce your issue on my computer. I'll fire up a Windows VM and check this issue during the course of the week. 

> Also this pop-up window overlays all other windows and does not disappear if another app is brought into focus. The same is true if any of the other graph or group windows are popped up by double-clicking on them. If the user goes to another program the open Serial Studio pop-ups overlay the other program, which should not occur. Open dialogues should only overlay the Serial Studio display.

The idea of having a "dialogs stay on top" was part of the initial design of SS. However, I never took into account that users would be using another app while using SS (SS was initially designed to work as a ground station software, in which no other apps should be running in the ground station computer during a mission/launch). 

I'll update the window flags accordingly so that dialogs do not inhibit the user for working with other apps. I'll keep you updated in this thread.

TDStrasser, 2021-02-09:
Thanks for the quick reply. I have another observation from using the new version.  The console record is not continuous when it is not being displayed and you have the dashboard showing instead.  See the attached screenshot. The data was coming in once per second.
[image]
[Serial Studio Console.txt](https://github.com/Serial-Studio/Serial-Studio/files/5948277/Serial.Studio.Console.txt)

alex-spataru, 2021-02-09:
> The console record is not continuous when it is not being displayed and you have the dashboard showing instead.

This is a temporary workaround to avoid slowing down the application at high data transfer speeds. The culprit of this was the control that I used to display console text, this workaround will be disabled in the next version (and console output will be logged even if the console is not visible).

alex-spataru, 2021-02-14:
Hi, I just released a [new version](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.16) that fixes the console issues that you mentioned, I also updated the window flags for the groups, graphs and widget windows. Any feedback is greatly appreciated!

TDStrasser, 2021-02-14:
Thanks Alex. The console continuity now appears to have been fixed.
The graph windows also now do not stay on top when other programs are activated. However the console player command dialogue still stays at the front regardless of program.  This is on Windows. I haven't tried the Mac version.
Thanks for your prompt attention to my reports.

alex-spataru, 2021-02-14:
This is planned, I left the CSV player window as a top-sided window due to several requirements in the projects in which I use SS. A possible workaround for this would be to integrate the player controls in the toolbar in a future release, so that users don't get carried away thinking that the CSV replay feature is showing data in realtime.

TDStrasser, 2021-02-14:
OK then that's fine if it is intentional for other use cases.  No need to change it then.
Thanks again.

closed 2021-02-16 by alex-spataru