Serial-Studio/Serial-Studio #24: "data to text" interpreter support

TCRobotics, 2021-02-05:
First of all, your project is the definitive serial console! Awesome work! 👏
I´m tired of realterm´s  errors and lack of advanced serial console alternatives 😭

I use finite state machines in my projects all the time, and the most important data is the current state.
Sending over serial the name as ascii is time consuming in time critical projects.

Maybe you can make a "data to text" interpreter, so you config an array of text at JSON file and the Serial Studio translates a number (index) to the stored text (JSON field).

Example:
Data frame
```/*4,1*/```
JSON
```json
{
   "t":"Minimal Example",
   "g":[
      {
         "t":"MCU Status",
         "d":[
            {
               "t":"State",
               "v":"%1",
               "text":[ "UNARMED", "ARMED", "FIRST STAGE", "SECOND STAGE", "FALLING", "REST" ]
            },
            {
               "t":"Parachute deployed",
               "v":"%2",
               "text":[ "NO", "YES" ]
            }
         ]
      }
   ]
}
```

I think this "data to text" feature could be usefull for many applications.
Maybe also with a text widget 😊

alex-spataru, 2021-02-06:
I think that is very good idea! I'll begin implementing it for the next release cycle (1.0.15, I am just doing final verifications for 1.0.14).

Suggestion, instead of `text`, I would use a key name like `enum`. I think that `enum` could help users infer faster how your idea works. 

I should also allow using keys like `title`, `value` and `data`. I feel that using only one character will eventually result in very confusing JSON files.

alex-spataru, 2024-10-31:
Hi! Apologies for the delayed response, and thank you for your patience. I wanted to follow up in case you're still interested in this feature. I strive to balance feature requests with the stability and usability of the program, so sometimes it takes a bit of time (and feedback) to find a robust and user-friendly solution for all of us.

The [frame parser function](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view) in the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.0.0) might be helpful here. Acting as a “man in the middle” between the data received from the microcontroller and what’s displayed on the dashboard, it enables you to customize the data output using JavaScript. For example, you can use a switch statement to display specific text values based on the numerical inputs/values from the microcontroller and then set up your JSON project in the Project Editor accordingly.

I'll close this issue for now to keep the repository organized. Please feel free to reopen it if needed or reach out with further questions. Thanks again for your input and interest!

closed 2024-10-31 by alex-spataru