Serial-Studio/Serial-Studio #25: Send Commands from the Dashboard

Genisis300, 2021-02-05:
I'm just asking here before i go to far into the code on this. I was wondering if it's possible to have some basic functions on the dashboard such as buttons / sliders to adjust some setpoints  / start stop a machine. 

Currently I'm thinking that some extra tags could be added to the JSON with a simple JSON string fired back over the serial on click.  I'm willing to take a look at adding this I'm just getting my head round QT at the Moment :)

DanielPrjct, 2021-02-08:
I totally agree with Genesis300 suggestions on Serial-Studio enhancement, adding these kind of controls.

Having buttons / sliders and/or text fields as widgets to send serial data to the MCUs will allow use SS as a "quasi-SCADA" program for low end microcontrollers.

I have developed OSIMPLC, an Open Hardware + Free Software project for practices on automation at technical secondary shcools (high schools) and professional training centers (see https://osimplc.com and https://github.com/osimplc/osimplc-docs ).

It will be a great improvement (for OSIMPLC, Arduino, Atmega, PIC and other 8 bits MCU users) if SS will provide these kind of widgets. Of course, JSON file must necessarily reside on the PC side, and MCUs must only send and receive serial data as .csv...

Sadly, I know nothing about Qt or software programming, so I cann't collaborate with SS developers...
But perhaps I can contribute as beta tester, or on other tasks.

Best regards,
Daniel

Genisis300, 2021-02-08:
i'm hoping to get some time soon i'm working on a couple of projects at the moment. But i have a few ideas that i'm hoping to add. 
another small improvement would be to store the json map files in the same folder as SS and have it load them if the project title from the MCU is found as a map file on the machine running SS.

I will say so far i'm really enjoying playing with it. we have it monitor a High pressure / gas system PID controller at the moment.

alex-spataru, 2024-11-17:
Implemented since version 3.0...closing this issue :)

closed 2024-11-17 by alex-spataru