Serial-Studio/Serial-Studio #27: Cannot set 3000000 baud

willtoth, 2021-02-08:
**Describe the bug**
When I manually type in 3000000 for the baud rate, the output is not correct. I can confirm in other terminal programs that the data is received correctly.

[image]

alex-spataru, 2021-02-08:
Hi, could you please send me your application log to check if the serial port module reports an error? You can find the application logs in `About` ➝ `Open log file`:

[image]

Thanks for the report!

alex-spataru, 2021-02-10:
In my case, I cannot set to 3000000 baud because the MCUs that I have do not support such speeds. Here is a screenshot of the output in CoolTerm:

[image]

And a `qDebug` printout of the data received directly from the serial port (before its added to the UI console display):

[image]

And regarding the overlapped characters in your screenshot, this is currently being fixed. In SS 1.0.14 and 1.0.15 I tried to implement my own text editor widget because the QML text editor offered by Qt was too slow for logging.

However, I am now using a QPlainTextEditor and integrating it to the QML interface, which solves the performance issue and avoids these kinds of UI glitches.

closed 2021-09-04 by alex-spataru