Serial-Studio/Serial-Studio #28: Duplicate Channels

willtoth, 2021-02-08:
**Is your feature request related to a problem? Please describe.**
My project includes many channels, where each channel includes multiple of the same measurement. For example:

Channel 1:
 - Voltage
 - Current
 - Speed

Channel 2:
 - Voltage
 - Current
 - Speed

Channel 3:
 - Voltage
 - Current
 - Speed

Currently, I must define the JSON to include each of these channels, and track their channel numbers. Instead, allowing channels to be defined, by defining how one channel behaves, and an identifier for each channel, you could scale to any number of identical channels.

Maybe some format like:
```
/*MyChannel[1], data, data, data*\
/*MyChannel[2], data, data, data*\
/*MyChannel[3], data, data, data*\
```

or

```
/*MyChannel[1], data, data, data,
MyChannel[2], data, data, data,
MyChannel[3], data, data, data*\
```

willtoth, 2021-02-08:
Introducing a concept of channels like this would also allow for more interesting grouping for the UI.

alex-spataru, 2021-02-09:
I am also currently thinking of a way to implement this feature, because I need to process both Rover and CanSat telemetry from the same ground station software (and I plan to use SS for this obviously).

For the moment, I was thinking of creating a JSON structure similar to:

```
{
   "title":"Project title",
   "channels":[
      {
         "header":"CH_1",
         "header_index":1,
         "title":"Channel 1",
         "groups":[
            {
               "title":"Channel 1 Group 1",
               "data":[
                  ...
               ]
            }
         ]
      },
      {
         "header":"CH_2",
         "header_index":1,
         "title":"Channel 2",
         "groups":[
            {
               "title":"Channel 2 Group 1",
               "data":[
                  ...
               ]
            }
         ]
      }
   ]
}
```

This would allow us to differentiate between channels by checking the value of a given `index`. For example, the MCU frames would be similar to:

**Channel 1**: `/*CH_1, data1, data2, data3,...*/`
**Channel 2**: `/*CH_2, data1, data2, data3,...*/`

*Note:* in the JSON above I use more descriptive keys, which are not valid for the moment. However, I think that changing `t` to `title`, `v` to `value`, etc. will be necessary to avoid issues as the JSON structure increases in complexity.

willtoth, 2021-02-09:
I think its a start, and it definitely makes it way easier than the existing schema, but one thought here is you still have to define exactly what data is in each channel, and if its the same set of data this creates a lot of repetition. For your example, say you have 20 identical CanSat streams coming in from the same platform. Instead of creating a new channel definition and copy-pasting for each one, create a way to differentiate channels with a single definition.

Maybe allow wildcards or something like that in the header definition?

Or force a naming convention in the data, such as ChannelName[UniqueId]

Channel 1: `/*CH[1], data1, data2, data3,...*/`
Channel 2: `/*CH[2], data1, data2, data3,...*/`
Channel 3: `/*SomeOtherChannel, data1, data2, data3,...*/`

```
{
   "title":"Project title",
   "channels":[
      {
         "header":"CH",
         "header_index":1,
         "title":"Channel %1",
         "groups":[
            {
               "title":"Channel %1 Group 1",
               "data":[
                  ...
               ]
            }
         ]
      },
      {
         "header":"SomeOtherChannel",
         "header_index":1,
         "title":"Some Other Data",
         "groups":[
            {
               "title":"SomeOtherChannel Group 1",
               "data":[
                  ...
               ]
            }
         ]
      }
   ]
}
```

chippey, 2021-04-02:
Piping in with a thought:
It would be nice if channels could be looked up by something like a Sensor ID.  I'm on a project right now where we're using many (dozens) of DS18B20 temperature sensors, and they each have a 64 bit unique ID (an example being '28AABD604C1401BD').  It would be great if that SensorID could be a key to log associated data (in my example, temperature).

Touching on another feature request that's in a different issue thread, it would be great if those 'channels' of data could live on the same widget (ie, for my case I would want each SensorID to be a line plotting on a single graph pane)

The format the data is streamed out from the logging platform is <monotonic_index>,<SensorID>,<Temperature>  so the concept of a channel or some other way of specifying a field in a stream of data as a key would be great!

daniel-wuyuezhou, 2021-11-16:
Inlcude duplicate channels, I think the duplicate frame index is very important. Because it's very nice to plots all the ADC channels together in one graph and meanwihle every ADC channel separately. Thank you.

alex-spataru, 2024-11-17:
Took me a while, but you can now to this since version 3.0, you can duplicate datasets without any issues in the [Project Editor](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor), and you can use the [frame parser](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view) for more advanced functionality.

closed 2024-11-17 by alex-spataru