Serial-Studio/Serial-Studio #29: Math expressions in manual JSON mode

willtoth, 2021-02-08:
For a resource limited MCU, sending raw data from a sensor may be the only feasible way to send back to the application, however the visualization and datalogging may want to post process it. For example, the raw register values from an accelerometer or similar, before calculating actual acceleration. Allowing the value (or a new key type) in the JSON be a math expression would allow any amount of post processing. 

Example:

```
{
         "t":"Accelerometer",
         "w":"accelerometer",
         "d":[
            {
               "t":"X",
               "v":"(%15 * 100) / 9.8",
               "u":"m/s^2",
               "g":true,
               "w":"x"
            },

```

**Describe the solution you'd like**
There seems to be some good libraries for this type of thing already, for example https://github.com/ArashPartow/exprtk

alex-spataru, 2021-02-09:
Hi, I like this idea! I think that we can use Qt directly for this using the [QJSEngine](https://doc.qt.io/qt-5/qjsengine.html) class, any valid JS code should be able to execute. It may be slower than [exprtk](https://github.com/ArashPartow/exprtk), but we won't need any additional dependencies to implement this feature.

alex-spataru, 2021-02-10:
I just implemented this feature. You can now use JS code in the `v` field of the JSON map, including functions like `Math.sin()`, `Math.cos()`, `Math.abs()`, etc.

The JSON frame and javascript expressions are evaluated in a different thread, so it should not affect the performance of the application.

alex-spataru, 2021-02-10:
And here is an example of my test case. I generate sine, cosine and abs. cos using the frame count and javascript code stored in the JSON map:

[image]

Relevant JSON map code:

```
{
   "t":"MCU Status",
   "d":[
      {
         "t":"Runtime",
         "v":"%1",
         "u":"ms"
      },
      {
         "t":"Frame Count",
         "v":"%2"
      }
   ]
},
{
   "t":"Function generator",
   "d":[
      {
         "t":"Sine",
         "v":"Math.sin(%2 * Math.PI / 180)",
         "g":true
      },
      {
         "t":"Cosine",
         "v":"Math.cos(%2 * Math.PI / 180)",
         "g":true
      },
      {
         "t":"Cosine (absolute)",
         "v":"Math.abs(Math.cos(%2 * Math.PI / 180))",
         "g":true
      }
   ]
}
```

alex-spataru, 2021-02-14:
Hi, I just released a [new version](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.16) that implements this suggestions. I would really appreciate your feedback in order to know if we should close this issue.

closed 2021-02-21 by alex-spataru

alex-spataru, 2021-12-05:
Hi, just to notify you that I needed to remove this feature to fix some issues (see #84, #85 and #86). I really think that its a nice feature to have, however, this can also be done in the post-processing phase when you already have your CSV file. If Serial Studio slows down or crashes while obtaining your data, it's no use to have so many features.