Serial-Studio/Serial-Studio #31: Customizable start and stop frames

willtoth, 2021-02-10:
Simply a mechanism to define some other token as the start or end of frame. One example of where this could be nice would be an existing stream that is going straight to CSV already, having '\r' or ';' for end of frame and '\n' for start of frame would mean it just works in both.

alex-spataru, 2021-02-11:
I just implemented this feature, users will now be able to change start/end sequences from the setup pane (see screenshot below):

[image]

As with other issues, I'll notify you when the new version is released to confirm changes before closing this issue.

jakkra, 2021-02-16:
Hi awesome feature! I tried it but I cannot get it to "detect" my data.

This is the format of my output where %d is replaced by integer values:
```
%d,%d,%d,%d,%d\r\n
%d,%d,%d,%d,%d\r\n
%d,%d,%d,%d,%d\r\n
...
```

My ideas was to make \r the end sequence and \n the start sequence. Can the input fields hande escaped characters such as `\r`?

alex-spataru, 2021-02-25:
> It works, had a error in json file (I output more variables than defined in the JSON).

Thanks a lot for the feedback! I will close this issue, if anyone encounters a problem regarding this feature, feel free to re-open this issue or create a new one.

closed 2021-02-25 by alex-spataru