Serial-Studio/Serial-Studio #33: More than one variable plotted on a graph

MajicDesigns, 2021-02-17:
Firstly thanks for a great piece of software that is making my life a lot easier. 

**Is your feature request related to a problem? Please describe.**
I am using this to tune a PID loop. To make this more effective it would be great to be able to combine more than one line on a graph. For example, having the Set Point and Current Value combined on the same plot would make it easier to see how quickly the process is settling down.

**Describe the solution you'd like**
Some way to say that two variable should be plotted together.

**Describe alternatives you've considered**
Currently showing the graphs side by side but sometimes the Y axis is scaled differently and makes it hard to compare.

alex-spataru, 2021-10-22:
I will close this issue due to lack of responses and to keep things clean. Feel free to re-open this issue if you encounter problems :)

MajicDesigns, 2021-10-22:
Will check this when I get the chance. Thanks for including the enhancement.