Serial-Studio/Serial-Studio #34: [Feature Request] MQTT Topic as network source

fusedFET, 2021-02-17:
With the rise of IoT many data are already sent back to MQTT on clouds.
I have some devices where all serial lines are already used, or devices are in the field. 
(And I cant use Segger RTT/Semihosting as source)
This means instead of going through a Serial, it would be easy to send the data to MQTT topic. 
(Bascially you could monitor anything on a cloud service with this approach in real-time).

**Describe the solution you'd like**
Adding to TCP/UDP drop-down an MQTT option.
Following fields would be required:
- Host
- MQTT Topic
- Authentification mode: Basic (User+PW) Certificates (Private Key Certificate, Certificate Authority)
- MQTT Name
Each MQTT message received is parsed, like a frame over serial.

(Yes, Serial over MQTT)

**Describe alternatives you've considered**
I tried many hacks to visualize data from my devices on the cloud, but those are all quite complicated architectures.

**Additional context**
I am using AWS IoT as backend, but any kind of MQTT broker could benefit from this.
Guess you could connect far more than devices to it, like whole cloud related metrics.

It's a pleasure to use Serial Studio. There simply is no solution which si so simple to use.
Thank you <3

alex-spataru, 2021-02-25:
Hi, 

This feature will be eventually implemented, I need to get it working for the CanSat competition because they require us to use MQTT to send incoming data over to their server. For the moment, I am currently playing with [qtmqtt](https://github.com/qt/qtmqtt) and learning more about MQTT in general.

I'll post updates on my progress on this thread. Thank you for the idea!

fusedFET, 2021-02-25:
That sound's great!

I have never used qtmqtt before, but it seems to be the most logical choice to keep a unifrom library stack. 

Just a proposition:
Serial Studio, when opened opens a defined python file as sub-process.
Each time a new data-set arrives into serial studio, the data is split to the different graph widgets, but also to 
a special handler function (per-widget) in this python file with the new data as argument in JSON format.

This would allow to simply process and republish data received on serial studio to some MQTT endpoint.
It might be simpler than changing the GUI to support generating MQTT frames.
At least in python it is quite simple to generate those frame, even more if some specific format is required by your competition :)
(Beside the fact that you could maybe even use this to make some automated testing with serial studio ?)

If I can help you somehow with scripts or, some tests on a cloud architecture or similar feel free to ping me :)

alex-spataru, 2021-02-25:
**Update:** I just [forked](https://github.com/alex-spataru/qtmqtt) the [Qt-MQTT project](https://github.com/qt/qtmqtt) so that I can integrate it with Serial Studio using sub-modules. 

> Serial Studio, when opened opens a defined python file as sub-process.
Each time a new data-set arrives into serial studio, the data is split to the different graph widgets, but also to
a special handler function (per-widget) in this python file with the new data as argument in JSON format.

This could be (relatively) easily done with [QProcess](https://doc.qt.io/qt-5/qprocess.html), and we could give users more power with what to do with incoming data. I will take your proposal into consideration. I think that letting users modify the Python script to suit their own needs could be very useful. 

I am currently in a pretty tight deadline to get this to work, and I am a lot more familiar with Qt/C++ than with Python. Also, the `qt-mqtt` library has some [examples](https://github.com/qt/qtmqtt/tree/dev/examples/mqtt) that I can use to get an idea on how to implement support for MQTT in Serial Studio.

The course of action that I am considering is:
- Integrate the Qt MQTT library and get it to work before I reach the deadline.
- Add support for calling Python scripts and pass the received data to these scripts to allow users to extend the functionality of Serial Studio. This can be related to the ideas discussed in #17.

alex-spataru, 2021-02-25:
**Edit:** Unfortunately, I cannot use Qt MQTT due to licensing issues. AFAIK, I cannot use GPLv3 code in a MIT-licensed project. And I personally feel GPLv3 too restrictive for the usage cases that users may have for this project.

alex-spataru, 2021-03-01:
**Update:** with the latest commit, Serial Studio supports sending RX data to a MQTT server/broker in CSV format. Here is a screenshot with the MQTT server for the CanSat competition:

[image]

The next step is to allow Serial Studio to act as a MQTT subscriber. This would also allow other instances of Serial Studio in the network/world to visualize the data received and processed by a SS ground station.

Finally, here is a screenshot of the MQTT settings pane:

[image]

alex-spataru, 2021-03-11:
I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.19) that allows Serial Studio TO send/receive data over MQTT. RX data is sent in a CSV-like format, and other instances of Serial Studio can display the broadcasted data using the same `*.json` file (you just need to set the "mode" combobox to "Subscriber"). 

Please let me know how it works for you. Greetings!

fusedFET, 2021-03-23:
That looks great 👍, but do you plan to support MQTT key/cert authentication for MQTT? 
I have some issues to add the classic password authentication to my brokers.

Good luck for the can-sat competition 👍

alex-spataru, 2021-03-24:
> But do you plan to support MQTT key/cert authentication for MQTT?

[Taking a look at the headers](https://github.com/alex-spataru/qmqtt/blob/37d55eecb706a5a45b09314e35263f01b6b5cd1f/src/mqtt/qmqtt_client.h#L160) of the library that I am using for MQTT communications, I think that is possible. I need to do some testing, but this feature will be eventually implemented.

> Good luck for the can-sat competition 👍

Thanks a lot! I will go back to working on SS as soon as I have some free time 😄

bmalcheski, 2021-10-29:
Any chance you'll be adding TLS encryption w/ and w/o certs?

Any intention to add further MQTT configuration options such as QoS, last will, keep alive, etc?

alex-spataru, 2021-11-01:
Hi, 

The MQTT library that I use for Serial Studio [has support for SSL configuration](https://github.com/alex-spataru/qmqtt/blob/868bb025c355b7e64b9a1ad3ea07c1a34768d5b1/src/mqtt/qmqtt_client.h#L153) with and without certificates, QoS, will retain and keep alive. I just need to add the UI controls to interface with the functions of the library.

Do you know of any broker that I can use to test these features? I don't know if HiveMQ allows me to test these configurations...

bmalcheski, 2021-11-01:
So no TLS support though? I looked through the header files you mentioned previously and didn't see anything about it TLS support myself. I bring it up because I was going to use SS to help me do some debug logging but can't if it doesn't support TLS. You may already be aware, but using MQTT with AWS requires SSL with certs. The issue I've run into is many edge devices I've used don't have the ability to use cert files hence encryption w/o authentication. 

I would think that HiveMQ supports those features. They have a bunch of articles regarding them... Otherwise, Mosquitto is another option. Obviously, you can't change the configs for either of the publicly hosted test servers but you can run your own instance to manipulate the broker config regarding these features.

https://www.hivemq.com/downloads/
https://mosquitto.org/download/

This looks like a great project and a very useful tool!

alex-spataru, 2021-11-03:
Thanks a lot for the links! I'll work on this issue during the weekend. I am currently on exam period. Regarding TLS, I *believe* that the [QSslSocket](https://doc.qt.io/qt-5/qsslsocket.html) class supports it (and thus QMQTT with SSL compilation flags enabled).

alex-spataru, 2021-11-12:
Hi! I am currently implementing the requested features. Advanced MQTT options & SSL/TLS config can be set in a separate dialog that can be shown from the MQTT tab. Here is a screenshot of the dialog:

[image]

For the moment, I have finished adding support for QOS, Keep-Alive and retain flags. SSL/TLS support is underway. Please let me know if I should change anything or if I forgot to add something to the MQTT config. dialog.

alex-spataru, 2021-12-05:
**Update:** I can confirm that the TLS/SSL feature works correctly with the latest commit. I will draft a new version in the upcoming days :)

alex-spataru, 2021-12-07:
Hi, just to let you know that I just released a [new version](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) that adds SSL/TLS support for MQTT connections. Please let me know how it works for you.

closed 2022-02-07 by alex-spataru