Serial-Studio/Serial-Studio #35: chart got worst in newer versions (after version 1.0.15)

Dzunior, 2021-02-17:
Hi,
I've made a design where I'm sending values (sine waveform) to Serial Studio. Version 1.0.15 looks nice, but when I updated to 1.0.16 and now to 1.0.17 I've noticed that the waveform looks bad. 

The design is the same (same JSON, same port settings). 

Version 1.0.15
[image]

Newer versions
[image]

Here's my JSON file I use with both versions.

[serial_studio.json](https://github.com/Serial-Studio/Serial-Studio/files/5998182/serial_studio.json.log)

Are there some changes I should make for newer versions?

Best regards,
Dominik

alex-spataru, 2021-02-17:
Hi, in version 1.0.16 I moved the serial data interpretation code to another thread to improve performance and avoid overloading the UI thread if the baud rate is too high. 

However, some serial frames are interpreted out of order. In version 1.0.17 I fixed this issue in the CSV export module, however, I did not update the UI modules. I'll fix this issue for the 1.0.18 release.

Thanks for the report!

Dzunior, 2021-02-17:
Great! Looking forward :)
BTW Great tool ! :)

alex-spataru, 2021-02-18:
For some reason GitHub automatically closed this issue with my commit message. Sorry!
Just to test, can you please tell me the aprox. frequency at which you transmit your data to Serial Studio?

dominikdomanski, 2021-02-18:
I send data continuously, so the frequency will be limited by the baud rate (115200).

alex-spataru, 2021-02-21:
Hi, I just uploaded a new release that should fix this bug. If possible, please test it & let me know if I should close this issue.

Dzunior, 2021-02-21:
Hi, 
It works perfectly now. Thank you! :)

commit 56d9ce10 (2021-02-18): Fix #35 & improve performance
closed 2021-02-21 by Dzunior