Serial-Studio/Serial-Studio #44: Compass widget

nono313, 2021-03-03:
**Is your feature request related to a problem? Please describe.**
Working with a magnetometer, I would like to be able to visualize the current direction on the device using a compass widget on the Serial Studio UI.

**Describe the solution you'd like**
A compass widget (round that turns to show the device's direction compared to the North).

**Describe alternatives you've considered**
I might be able to cheat and use the gyroscope widget but it wouldn't really look nice (no indication of North).

**Additional context**

alex-spataru, 2021-03-12:
Hi, sorry for the late reply. I was kind of busy working on other projects. I just implemented this feature with the latest commit (see screenshot below), it should be available with the next release (10.0.20) of Serial Studio.

[image]

alex-spataru, 2021-03-12:
To use it, add `compass` in the `w` field of any dataset. The value field should be an integer in the range of `0` to `360`.

```json
{
   "t":"Compass",
   "v":"%1",
   "w":"compass"
 }
```

alex-spataru, 2021-03-14:
**Update:** I just released a new version that implements the compass widget. If possible, please try it and let me know how it works for you.

nono313, 2021-03-15:
Thank you very much for the speed of your response. This will be very helpful for me.

One suggestion : could it be possible to add an option (or maybe a second widget) to have the plane rotate instead of the graduations ? For my use case I am using the angle to light up LEDs on a circle of the device to indicate a direction. If we could have the plane image rotate and show the direction then we could easily compare with what the LEDs indicate on our device.

alex-spataru, 2021-03-16:
> Could it be possible to add an option (or maybe a second widget) to have the plane rotate instead of the graduations?

Of course, I will do some modifications on the widget during the course of the week & send some screenshots of the changes in this thread.

n8maher, 2021-06-06:
It would also be really cool to overlay a arrow (or other image) onto the GPS map to indicate heading as well! Let me know if this should be broken out as a separate enhancement request.

alex-spataru, 2021-09-04:
*Update:* I just created a new [release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.21) that implements this widget. Documentation for the compass widget is available [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Introduction-to-widgets#compass). 

> could it be possible to add an option (or maybe a second widget) to have the plane rotate instead of the graduations ?

> It would also be really cool to overlay a arrow (or other image) onto the GPS map to indicate heading as well! Let me know if this should be broken out as a separate enhancement request.

I will work on these requests for the next release. I know that its been some time, but I'll eventually get these issues addressed.

alex-spataru, 2021-09-26:
> One suggestion : could it be possible to add an option (or maybe a second widget) to have the plane rotate instead of the graduations ? For my use case I am using the angle to light up LEDs on a [...]

Hi, I am currently working on moving all the widgets & plots to [Qwt](https://qwt.sourceforge.io). This comes with two benefits:
a) All the widgets are implemented in C++, thus improving application performance.
b) Widgets can now be implemented with less hassle than before, since I do all the widget loading in a single class.

For the moment, I am still getting my head around Qwt, however, one of the first widgets that I implemented is the compass. Here is a preliminary screenshot of the new compass widgets (subject to change in the nexts commits 😆):

[image]

alex-spataru, 2021-09-26:
And here are some screenshots of compass widgets build with Qwt (from their own documentation):

[image]

alex-spataru, 2021-10-20:
I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.0) with a new compass design. Please let me know if I need to change anything.

alex-spataru, 2021-10-22:
I will close this issue due to lack of responses and to keep things clean. If anyone encounters problems, feel free to re-open this issue.

closed 2021-10-22 by alex-spataru