Serial-Studio/Serial-Studio #47: Add support for binary sensors/flags

sixstringsg, 2021-03-20:
**Is your feature request related to a problem? Please describe.**
Not to a specific problem, just a missing feature

**Describe the solution you'd like**
In addition to the strings that are already supported, I would love to be able to pass a boolean value and have an on/off indicator. This could be used to indicate that a value is out of the expected range, could be used to monitor binary sensors, and could also be used to monitor the state of relays. 

**Describe alternatives you've considered**
In a similar vein, it would be incredible to be able to add a indicator bar on the bar/tank widget, for a similar "value is out of nominal readings" function, similar to how the redline section on a car's tachometer functions

**Additional context**
At the basic level, I'm looking for something similar to how [this project](https://github.com/mladenilic/arduino-cereal) handles bools, it would be great if they could be adjacent to a graph or a value, but their own widget would work fine as well!

alex-spataru, 2021-03-22:
Hi! Thanks for the suggestions, I will begin working on them as soon as I have some free time. I'll keep you updated through this thread.

sixstringsg, 2021-03-23:
Awesome, thanks so much! I'm building an engine monitoring system for an offroad race truck that I'm planning on using this project to display vital stats to support vehicles.

zhigangbox, 2021-04-26:
Not only 0 / 1 indication, but also 0 / 1 setting

alex-spataru, 2021-10-22:
> In a similar vein, it would be incredible to be able to add a indicator bar on the bar/tank widget, for a similar "value is out of nominal readings" function, similar to how the redline section on a car's tachometer functions

With the latest release, Serial Studio allows users to set an "alarm level" for bar widgets directly with the JSON editor. I look forward to implement the boolean indicators for the next release (1.1.1).

sixstringsg, 2021-12-06:
How does one actually set the LED to on or off? Is it just expecting a binary signal of 0 for on or 1 for off?

alex-spataru, 2021-12-06:
> How does one actually set the LED to on or off? Is it just expecting a binary signal of 0 for on or 1 for off?

Hi, if the absolute value of the signal is less than 0.1, then the LED shall be turned off. Otherwise, the LED shall be turned on.

Here is a an abstract of the code, full implementation [here](https://github.com/Serial-Studio/Serial-Studio/blob/dcf30358de7ec6bec7626fb68900809b3628d6a2/src/Widgets/LEDPanel.cpp#L176):

```cpp
if (qAbs(value) < 0.10)
    m_leds.at(i)->off();
else
    m_leds.at(i)->on();
```

We check turn the LED off at `0.1` to deal with noisy ADCs connected to ground.