Serial-Studio/Serial-Studio #49: Add graph display mode or a new widget similar to spectrom analyser

matrixant, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
I want my data to appear on a graph or widget like a spectrum, and the graph there will only be updated when I send new data.

Like this:
[image]

**Describe the solution you'd like**
- Maybe add a mode parameter (optional) and index data field (optional) to graph, and then the graph would only update the value of the newly arrived correspond to the index.

eg. 
```
    {
        "t": "Spectrum",
        _"m": "indexed"_,
        _"i":"%1"_,
        "v": "%2",
        "g": true
    }
```

- Or add a new widget

eg.
```
{
    "t": "Spectrum",
    _"w": "spectrum"_,
    "d": [
            {
               "t": "Level",
               "v": "%2",
               "u": "dBW",
               "g": true,
               _"w": "value"_,
               "min": 0,
               "max": -150
            },
            {
                "t": "Freq",
                "v": "%1",
                "u": "kHz",
                "g": true,
                _"w": "index"_,
                "min": 0,
                "max": 50000
            }
        ]
    }
```

alex-spataru, 2021-04-20:
Hi! Sorry for the late response. This issue is closely related to #5. I'll close the first issue & continue the discussion here.

This is the original message posted by @fusedFET:

> Hello,
> I have been using this tool to visualize in real time the effect of DSP on the device.
> (Which was very usefull by the way!)
> 
> ** Problem **
> If I have data which needs to be shown that is not in the time domain.
> For example: FFT/DFT which processed on the device or 1D cameras.
> 
> ** Solution **
> 2 Widgets where the Y axis is not time based.
> 
> A classic line-chart, based on a table.
> a heatmap/waterfall diagramm to show for example 1D grayscale diagrams in time or an FFT/DFT in time.
> The device would just send the row/column count followed by values set in one transmission.
> 
> Describe alternatives you've considered
> Exporting signal data to Scilab/Octave, which is tedious but is also usefull.

I am still researching how to do this. For the moment, I am considering to discard the `Qt Charts` module and use [Qwt](https://qwt.sourceforge.io) for all graphing/plotting functions. If you are interested, here is a [video](https://www.youtube.com/watch?v=YRIkYHixZtA) of a spectrum/waterfall plot being generated with Qwt. 

@fusedFET @matrixant Please let me know if the plot shown in the YT video is what you are looking for!

matrixant, 2021-04-20:
@alex-spataru This is what i need.

fusedFET, 2021-04-20:
@alex-spataru A widget as shown on the video would totally work for me :)

alex-spataru, 2021-10-09:
Hi, after more than half a year I just began implementing this widget. Here is a (preliminary) result:

[image]

The left graph represents the FFT transformation, while the right graph shows the data as it is received.

To create a FFT plot, users will need to select the "FFT plot" option and specify the number of samples to use in the JSON Editor window:

[image]

If you are interested, the code for the FFT Plot widget is defined [here](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/Widgets/FFTPlot.cpp). I'll notify you when I draft a new release that contains this feature.

alex-spataru, 2021-10-09:
And here is a screenshot with square waves and sine waves:

[image]

Please let me know if the FFT plots are correctly generated. I still haven't seen FFT-related stuff in my courses, so I might be doing something wrong.

alex-spataru, 2021-10-22:
I will close this issue due to lack of responses and to keep things clean. If anyone encounters problems with the FFT plot, please re-open this issue.

closed 2021-10-22 by alex-spataru