Serial-Studio/Serial-Studio #55: logarithmic scales :) ?

3mrrrx, 2021-04-24:
logarithmic scales for plots would be good to have for some kinds of data/signals

alex-spataru, 2021-09-26:
Hi, I just integrated Serial Studio with [qwt](https://qwt.sourceforge.io/index.html) in the last commits. I am currently working on "porting" the previous QML/JS widgets to Qwt. Once I finish that (and familiarize myself with Qwt), I'll begin implanting more advanced graphs, including logarithmic scales and FFT/DFT plots (#49).

alex-spataru, 2021-10-08:
Hi, I just implemented this feature. Here is an example of the generated plot (with a rather unassertive dataset - altitude in meters):

[image]

To enable a logarithmic plot, you need to enable both the "generate plot" and "logarithmic plot" options in the JSON Editor window:

[image]

Please let me know what you think about this! I'll mention in this thread when a draft a new release containing this feature.

alex-spataru, 2021-10-20:
I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.0) that implements this feature. Please let me know how it works for you!

alex-spataru, 2021-10-22:
I will close this issue due to lack of responses and to keep things clean. If you encounter problems please feel free to open this issue again :)

closed 2021-10-22 by alex-spataru