Serial-Studio/Serial-Studio #56: 接收处理问题

GitCP, 2021-04-25:
在设置好115200 8N1 没有流控制的情况下，接收中文出现乱码

alex-spataru, 2021-04-26:
Hi! If you click on the "export console data" button, do the characters appear as they should on your system's text editor? I want to know if this issue is caused by the data display code or the serial data processing code.

[image]

Attached to this message is a screenshot with the location of the "export console data" button.

alex-spataru, 2021-09-04:
I'm closing this issue for now to focus on what should be done for the next release. Please let me know if I should re-open this issue :)

closed 2021-09-04 by alex-spataru