Serial-Studio/Serial-Studio #63: how to use auto mode to display widgets

wengchuang, 2021-06-30:
Hi:
I want use auto mode to display widgets.I send json datas to Serial-Studio,but the Serial-Studio can not display.
I send the json data like this:
/*{
   "t":"test",
   "g":[
      {
         "t":"Mission Status",
         "d":[
            {
               "t":"Runtime",
               "v":2,
               "u":"ms"
            },
            {
               "t":"Packet count",
               "v":3
            },
            {
               "t":"Battery voltage",
               "v":4,
               "g":true,
               "u":"V",
               "w":"bar",
               "min":3.6,
               "max":4.3
            }
         ]
      },
      {
         "t":"Sensor Readings",
         "d":[
            {
               "t":"Temperature",
               "v":5,
               "g":true,
               "u":"°C",
               "w":"bar",
               "min":0,
               "max":80
            },
            {
               "t":"Altitude",
               "v":6,
               "u":"m",
               "w":"bar",
               "min":0,
               "max":3000
            },
            {
               "t":"Pressure",
               "v":7,
               "u":"KPa",
               "g":true,
               "w":"bar",
               "min":54,
               "max":102
            },
            {
               "t":"External Temperature",
               "v":8,
               "g":true,
               "u":"°C",
               "w":"bar",
               "min":0,
               "max":80
            },
            {
               "t":"Humidity",
               "v":9,
               "g":true,
               "u":"%",
               "w":"bar",
               "min":0,
               "max":100
            }
         ]
      },
      {
         "t":"GPS",
         "w":"map",
         "d":[
            {
               "t":"GPS Time",
               "v":10
            },
            {
               "t":"Longitude",
               "v":11,
               "u":"°E",
               "w":"lon"
            },
            {
               "t":"Latitude",
               "v":12,
               "u":"°N",
               "w":"lat"
            },
            {
               "t":"Altitude",
               "v":13,
               "u":"m"
            },
            {
               "t":"No. Sats",
               "v":14
            }
         ]
      },
      {
         "t":"Accelerometer",
         "w":"accelerometer",
         "d":[
            {
               "t":"X",
               "v":15,
               "u":"m/s^2",
               "g":true,
               "w":"x"
            },
            {
               "t":"Y",
               "v":16,
               "u":"m/s^2",
               "g":true,
               "w":"y"
            },
            {
               "t":"Z",
               "v":17,
               "u":"m/s^2",
               "g":true,
               "w":"z"
            }
         ]
      },
      {
         "t":"Gyroscope",
         "w":"gyro",
         "d":[
            {
               "t":"X",
               "v":18,
               "u":"°",
               "g":true,
               "w":"yaw"
            },
            {
               "t":"Y",
               "v":19,
               "u":"°",
               "g":true,
               "w":"roll"
            },
            {
               "t":"Z",
               "v":20,
               "u":"°",
               "g":true,
               "w":"pitch"
            }
         ]
      }
   ]
}*/

Can you give me some suggest,Thanks very mutch!!!

fengzhizi32, 2021-06-30:
Are you sure it didn't go wrong?

wengchuang, 2021-06-30:
Thank you for you response. How to use Serial Studio in auto mode,can you give me a demo,Thanks!!!

alex-spataru, 2021-09-23:
Hi, please try using strings in the `v` field, for example:

```json
{
   "t":"Z",
   "v":"20",
   "u":"°",
   "g":true,
   "w":"pitch"
}
```

An example program for Arduino with your JSON file is:

```cpp
const char* json = "{\"t\":\"test\",\"g\":[{\"t\":\"Mission Status\",\"d\":[{\"t\":\"Runtime\",\"v\":\"2\",\"u\":\"ms\"},{\"t\":\"Packet count\",\"v\":\"3\"},{\"t\":\"Battery voltage\",\"v\":\"4\",\"g\":true,\"u\":\"V\",\"w\":\"bar\",\"min\":3.6,\"max\":4.3}]},{\"t\":\"Sensor Readings\",\"d\":[{\"t\":\"Temperature\",\"v\":\"5\",\"g\":true,\"u\":\"°C\",\"w\":\"bar\",\"min\":0,\"max\":80},{\"t\":\"Altitude\",\"v\":\"6\",\"u\":\"m\",\"w\":\"bar\",\"min\":0,\"max\":3000},{\"t\":\"Pressure\",\"v\":\"7\",\"u\":\"KPa\",\"g\":true,\"w\":\"bar\",\"min\":54,\"max\":102},{\"t\":\"External Temperature\",\"v\":\"8\",\"g\":true,\"u\":\"°C\",\"w\":\"bar\",\"min\":0,\"max\":80},{\"t\":\"Humidity\",\"v\":\"9\",\"g\":true,\"u\":\"%\",\"w\":\"bar\",\"min\":0,\"max\":100}]},{\"t\":\"GPS\",\"w\":\"map\",\"d\":[{\"t\":\"GPS Time\",\"v\":\"10\"},{\"t\":\"Longitude\",\"v\":\"11\",\"u\":\"°E\",\"w\":\"lon\"},{\"t\":\"Latitude\",\"v\":\"12\",\"u\":\"°N\",\"w\":\"lat\"},{\"t\":\"Altitude\",\"v\":\"13\",\"u\":\"m\"},{\"t\":\"No. Sats\",\"v\":\"14\"}]},{\"t\":\"Accelerometer\",\"w\":\"accelerometer\",\"d\":[{\"t\":\"X\",\"v\":\"15\",\"u\":\"m/s^2\",\"g\":true,\"w\":\"x\"},{\"t\":\"Y\",\"v\":\"16\",\"u\":\"m/s^2\",\"g\":true,\"w\":\"y\"},{\"t\":\"Z\",\"v\":\"17\",\"u\":\"m/s^2\",\"g\":true,\"w\":\"z\"}]},{\"t\":\"Gyroscope\",\"w\":\"gyro\",\"d\":[{\"t\":\"X\",\"v\":\"18\",\"u\":\"°\",\"g\":true,\"w\":\"yaw\"},{\"t\":\"Y\",\"v\":\"19\",\"u\":\"°\",\"g\":true,\"w\":\"roll\"},{\"t\":\"Z\",\"v\":\"20\",\"u\":\"°\",\"g\":true,\"w\":\"pitch\"}]}]}";

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.print("/*");
  Serial.print(json);
  Serial.print("*/");
  Serial.print("\n");
  delay(1000);
}
```

alex-spataru, 2021-09-23:
Finally, here is a program that reads all the ADC ports from an Arduino UNO and generates a JSON frame:

```cpp
///
/// JSON frame format (with %s in value fields)
///
const char* FORMAT = "{\"g\":[{\"d\":[{\"g\":true,\"max\":5,\"min\":0,\"t\":\"ADC 1\",\"u\":\"V\",\"v\":\"%s\",\"w\":\"bar\"},{\"g\":true,\"max\":5,\"min\":0,\"t\":\"ADC 2\",\"u\":\"V\",\"v\":\"%s\",\"w\":\"bar\"},{\"g\":true,\"max\":5,\"min\":0,\"t\":\"ADC 3\",\"u\":\"V\",\"v\":\"%s\",\"w\":\"bar\"},{\"g\":true,\"max\":5,\"min\":0,\"t\":\"ADC 4\",\"u\":\"V\",\"v\":\"%s\",\"w\":\"bar\"},{\"g\":true,\"max\":5,\"min\":0,\"t\":\"ADC 5\",\"u\":\"V\",\"v\":\"%s\",\"w\":\"bar\"}],\"t\":\"ADC Readings\",\"w\":\"\"}],\"t\":\"ADC Test\"}";

///
/// Strings to store ADC readings
///
char ADC1_STR[6];
char ADC2_STR[6];
char ADC3_STR[6];
char ADC4_STR[6];
char ADC5_STR[6];

/// 
/// String to store JSON frame
///
char JSON_DATA[512];

///
/// Reads voltage in the given @a pin and outputs it to the given @a str
///
inline void readAdc(const int pin, char* str) {
  const float voltage = analogRead(pin) * 5.0 / 1024;
  dtostrf(voltage, 4, 2, str);
}

///
/// MCU init function
///
void setup() {
  // Initialize serial port
  Serial.begin(115200);

  // Setup pins
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
}

///
/// MCU loop function
///
void loop() {
  // Read ADC values and store them into strings
  readAdc(A1, ADC1_STR);
  readAdc(A2, ADC2_STR);
  readAdc(A3, ADC3_STR);
  readAdc(A4, ADC4_STR);
  readAdc(A5, ADC5_STR);
  
  // Create JSON frame
  sprintf(JSON_DATA, FORMAT, ADC1_STR, ADC2_STR, ADC3_STR, ADC4_STR, ADC5_STR);

  // Send JSON data to serial port
  Serial.print("/*");
  Serial.print(JSON_DATA);
  Serial.print("*/");
  Serial.print("\n");

  // Wait 50 ms
  delay(50);
}
```

In this example, we need to use the `dtostrf()` function because `sprintf()` does not support `float` values in Arduino. As you may notice, *storing the JSON frame format string and converting every reading to a string is not very efficient* for small microcontrollers or for large projects. 

In that case, you might want to use the "Manual" mode:

JSON file created with the *built-in JSON editor*:

```json
{"fe":"","fs":"","g":[{"d":[{"g":true,"max":5,"min":0,"t":"ADC 1","u":"V","v":"%1","w":"bar"},{"g":true,"max":5,"min":0,"t":"ADC 2","u":"V","v":"%2","w":"bar"},{"g":true,"max":5,"min":0,"t":"ADC 3","u":"V","v":"%3","w":"bar"},{"g":true,"max":5,"min":0,"t":"ADC 4","u":"V","v":"%4","w":"bar"},{"g":true,"max":5,"min":0,"t":"ADC 5","u":"V","v":"%5","w":"bar"}],"t":"ADC Readings","w":""}],"s":"","t":"ADC Test"}
```

Arduino code:
```cpp
///
/// MCU init function
///
void setup() {
  // Initialize serial port
  Serial.begin(115200);

  // Setup pins
  pinMode(A1, INPUT);
  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
}

///
/// MCU loop function
///
void loop() {
  // Read ADC values
 const float adc1 = analogRead(A1) * 5.0 / 1024;
 const float adc2 = analogRead(A2) * 5.0 / 1024;
 const float adc3 = analogRead(A3) * 5.0 / 1024;
 const float adc4 = analogRead(A4) * 5.0 / 1024;
 const float adc5 = analogRead(A5) * 5.0 / 1024;
 
  // Send frame to serial port
  Serial.print("/*");
  Serial.print(adc1);
  Serial.print(",");
  Serial.print(adc2);
  Serial.print(",");
  Serial.print(adc3);
  Serial.print(",");
  Serial.print(adc4);
  Serial.print(",");
  Serial.print(adc5);
  Serial.print("*/");
  Serial.print("\n");

  // Wait 50 ms
  delay(50);
}
```

alex-spataru, 2021-10-22:
I will close this issue due to lack of responses and to keep things clean. If anyone encounters problems, feel free to re-open this issue.

closed 2021-10-22 by alex-spataru