Serial-Studio/Serial-Studio #64: gauge widget

Bischtinius, 2021-07-02:
First of all, thank you for this software. It helps a lot with my studies.

We made an RC-Car in university and I thought it would be nice to show the vehicle speed on a gauge.

I think a widget which is a gauge could be a nice feature 
Like in the “bar” widget the JSON could include “min” and “max” and the value

alex-spataru, 2021-09-03:
Hi, as of the latest commit you can now use gauge widgets within Serial Studio:

[image]

Here is the JSON code for widget usage:

```
{
   "t":"Title", 
   "v":"%value%,  
   "w":"gauge",  
   "min":%min_value%, 
   "max":%max_value%
}
```

I will notify you when I draft a new release containing this change. Please let me know what you think about it!

alex-spataru, 2021-09-04:
*Update:* I just created a new [release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.21) that implements this widget. Documentation for the gauge widget is available [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Introduction-to-widgets#gauge). Please let me know if I should close this issue.

Bischtinius, 2021-09-16:
I tested the gauge widget today and if I configure it in the JSON the "Widget" option in the UI is not active.
I manged to work around the problem with a bar widget and then change the JSON.
I can say that the Widget itself works perfectly but the UI button "Widget" stays inactive.
[image]

alex-spataru, 2021-09-17:
Hi @Bischtinius, I fixed that issue in commit 050f64e749db10c1e29a159c8283560d78387026. I will notify you when I draft a new release containing this fix.

Thanks for the report!

alex-spataru, 2021-09-19:
The issue should be fixed with the [latest](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.22) release. Please let me know if it works for you!

closed 2021-09-23 by alex-spataru

Frenkenstein6, 2021-09-26:
Looks Great