Serial-Studio/Serial-Studio #66: cannot show '0A'  '0D' in binary display mode

noodlefighter, 2021-08-04:
**Describe the bug**

cannot show '0A'  '0D' in binary display mode

**To Reproduce**
Steps to reproduce the behavior:
1. connect to USB-TTL device (like /dev/ttyUSB0), loop-back TXD and RXD
2. switch to binary display mode: Console - Display Mode - Binary(hexadecimal)
3. switch to binary display mode: Console - Echo user command
4. switch to hex input mode
5. send '01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f'
6. '0A', '0D' lost
  ```
17:34:24.738 -> 
17:40:08.531 -> 01 02 03 04 05 06 07 08  09 0B 0C 0E 0F           |  ............. 
17:40:08.531 -> 
17:40:08.546 -> 01 02 03 04 05 06 07 08  09 0B 0C 0E 0F           |  ............. 
  ```

**Expected behavior**
...

**Screenshots**

[image]
[image]

**Upload your log file**

```
[11:45:16.152] QDateTime(2021-08-04 11:45:16.152 CST Qt::LocalTime)                     [I] [main]
[11:45:16.152] Serial Studio 1.0.20                                                     [I] [main]
[11:45:16.152] Running on Manjaro Linux                                                 [I] [main]
[11:45:16.153] Initializing C++ modules                                                 [I] [ModuleManager::initializeQmlInterface]
[11:45:16.154] Baud rate set to 9600                                                    [I] [IO::DataSources::Serial::setBaudRate]
[11:45:16.154] Data bits set to QSerialPort::Data8                                      [I] [IO::DataSources::Serial::setDataBits]
[11:45:16.154] Stop bits set to QSerialPort::OneStop                                    [I] [IO::DataSources::Serial::setStopBits]
[11:45:16.154] Serial port parity set to QSerialPort::NoParity                          [I] [IO::DataSources::Serial::setParity]
[11:45:16.154] Flow control set to QSerialPort::NoFlowControl                           [I] [IO::DataSources::Serial::setFlowControl]
[11:45:16.161] Finished initializing C++ modules                                        [I] [ModuleManager::initializeQmlInterface]
[11:45:16.161] Loading QML interface...                                                 [I] [ModuleManager::initializeQmlInterface]
[11:45:16.762] qrc:/qml/SetupPanes/MQTT.qml:123:15: QML TextField: Binding loop detected for property "text" [W] []
[11:45:16.764] qrc:/qml/SetupPanes/Network.qml:107:15: QML TextField: Binding loop detected for property "text" [W] []
[11:45:17.063] Baud rate set to 9600                                                    [I] [IO::DataSources::Serial::setBaudRate]
[11:45:17.118] Finished loading QML interface                                           [I] [ModuleManager::initializeQmlInterface]
[11:45:43.976] Baud rate set to 115200                                                  [I] [IO::DataSources::Serial::setBaudRate]
[11:45:53.262] Serial port error QSerialPort::NoError                                   [I] [IO::DataSources::Serial::handleError]
[11:45:53.383] Device opened successfully                                               [I] [IO::Manager::connectDevice]
[11:52:47.916] Disconnected from "ttyUSB0"                                              [I] [IO::DataSources::Serial::disconnectDevice]
[11:52:47.920] Device disconnected                                                      [I] [IO::Manager::disconnectDevice]
[11:59:44.396] Serial port error QSerialPort::NoError                                   [I] [IO::DataSources::Serial::handleError]
[11:59:44.516] Device opened successfully                                               [I] [IO::Manager::connectDevice]
[12:00:28.049] Serial port error QSerialPort::ReadError                                 [I] [IO::DataSources::Serial::handleError]
[12:00:28.050] Disconnected from "ttyUSB0"                                              [I] [IO::DataSources::Serial::disconnectDevice]
[12:00:28.052] Device disconnected                                                      [I] [IO::Manager::disconnectDevice]
[12:02:30.769] Serial port error QSerialPort::NoError                                   [I] [IO::DataSources::Serial::handleError]
[12:02:30.892] Device opened successfully                                               [I] [IO::Manager::connectDevice]
[15:23:17.572] Disconnected from "ttyUSB0"                                              [I] [IO::DataSources::Serial::disconnectDevice]
[15:23:17.574] Device disconnected                                                      [I] [IO::Manager::disconnectDevice]
[15:23:18.049] Serial port error QSerialPort::NoError                                   [I] [IO::DataSources::Serial::handleError]
[15:23:18.058] Device opened successfully                                               [I] [IO::Manager::connectDevice]
[17:39:45.570] Serial port error QSerialPort::ReadError                                 [I] [IO::DataSources::Serial::handleError]
[17:39:45.570] Disconnected from "ttyUSB0"                                              [I] [IO::DataSources::Serial::disconnectDevice]
[17:39:45.575] Device disconnected                                                      [I] [IO::Manager::disconnectDevice]
[17:40:03.809] Serial port error QSerialPort::NoError                                   [I] [IO::DataSources::Serial::handleError]
[17:40:03.936] Device opened successfully                                               [I] [IO::Manager::connectDevice]
```

**Upload your JSON map file**
unused

alex-spataru, 2021-08-05:
Hi, this issue comes from the following [code](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/IO/Console.cpp#L657):

```cpp
QString Console::hexadecimalStr(const QByteArray &data)
{
    // Remove line breaks from data
    QByteArray copy = data;
    copy.replace("\n", "");
    copy.replace("\r", "");

    // Convert data to string with dump every ~80 chars
    QString str;
    const int characters = 80;
    for (int i = 0; i < copy.length(); ++i)
    {
        QByteArray line;
        for (int j = 0; j < qMin(characters, copy.length() - i); ++j)
            line.append(copy.at(i + j));

        i += characters;
        str.append(HexDump(line.data(), line.size()));
    }

    // Return string
    return str;
}
```

I removed line breaks (`\n` & `\r`) to avoid formatting issues with the text display widget. I'll update the code and notify updates here.

alex-spataru, 2021-09-03:
Hi, just to confirm that the latest commits seem to fix this issue without any regressions. I will notify you when I draft the new release.

[image]

alex-spataru, 2021-09-04:
*Update:* I just created a new [release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.21) that fixes this issue. Please let me know if I should close this issue.

noodlefighter, 2021-09-04:
i have tried the new release. seems that the problem is solved!! THX :+1:

alex-spataru, 2021-09-04:
You're welcome, thanks for the report!

closed 2021-09-04 by alex-spataru