Serial-Studio/Serial-Studio #67: A question about how to update the UI

senliontec, 2021-08-17:
In Generator.cpp, line 182:

emit jsonFileMapChanged();

It seems that the signal jsonFileMapChanged() is emitted, however, i can not find any function to connect with the signal jsonFileMapChanged() to update the UI.  Frustrating, the UI updated successfully. Could you give me some suggestions ?

Thanks.

Frenkenstein6, 2021-08-17:
What State are you in.?   What happens when you GitHub sex or porn?

alex-spataru, 2021-08-17:
@Frenkenstein6 please take a look at the [Code of Conduct](https://github.com/Serial-Studio/Serial-Studio/blob/master/CODE_OF_CONDUCT.md).

alex-spataru, 2021-08-17:
@senliontec The `jsonFileMapChanged()` signal is used to [update](https://github.com/Serial-Studio/Serial-Studio/blob/fcb1681746ea51aaab90e360108dec8d0f6b1cd6/src/JSON/Generator.h#L70) the following QML/C++ properties:
- [`jsonMapFilename`](https://github.com/Serial-Studio/Serial-Studio/search?q=jsonMapFilename)
- [`jsonMapFilepath`](https://github.com/Serial-Studio/Serial-Studio/search?q=jsonMapFilepath)
- [`operationMode`](https://github.com/Serial-Studio/Serial-Studio/search?q=operationMode)

Most of the UI signals are handled by the following classes:
- [`DataProvider`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/UI/DataProvider.h)
- [`GraphProvider`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/UI/GraphProvider.h)
- [`WidgetProvider`](https://github.com/Serial-Studio/Serial-Studio/blob/master/src/UI/WidgetProvider.h)

The "provider" classes use the information from the latest JSON frame to generate the user interface that displays the current state of the microcontroller project.

senliontec, 2021-08-18:
@alex-spataru Thanks a lot.

closed 2021-08-18 by senliontec