Serial-Studio/Serial-Studio #71: CSV export doesn't strip frame end characters

dwhitters, 2021-09-22:
**Describe the bug**
For a frame: /*Frame,1,2,3*/, in the CSV, the '3' will be recorded as "3*/".

**To Reproduce**
Steps to reproduce the behavior:
1. Allow CSV Exporting
2. Open the CSV

**Environment**
Ubuntu 21.04

**Expected behavior**
The frame end characters will not be present in the CSV.

**Screenshots**
[image]

alex-spataru, 2021-09-22:
Hi, I just fixed this issue. I'll draft a new release in a moment.

alex-spataru, 2021-09-23:
I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.23) that fixes this issue. Please let me know if it works for you too!

dwhitters, 2021-09-27:
The issue has been resolved, thanks for the quick turnaround!

commit 5085f13a (2021-09-22): Fix #71
closed 2021-09-27 by dwhitters