Serial-Studio/Serial-Studio #72: Auto-reconnect

jcomas, 2021-09-22:
**Is your feature request related to a problem? Please describe.**
Yes. Other programs as Tera Term reconnects automatically when you disconnect and connect an USB-Serial connection.

**Describe the solution you'd like**
Add a checkbox called "auto-reconnect"

**Describe alternatives you've considered**
I use Tera Term program because it has auto-reconnect.

**Additional context**
Checkbox added to below of Serial options.

alex-spataru, 2021-09-23:
Hi, thanks for the idea! Here's a screenshot of how this feature would look like: 

[image]

I'll begin working on the implementation during the course of the week.

jcomas, 2021-10-20:
I'm testing release 1.1.0 with auto-reconnect feature. This version doesn't show any communication. It only works if you use other program to open/close serial communication and then you back to Serial Studio and you click "Connect" button again.

alex-spataru, 2021-10-20:
Hi, unfortunately I cannot replicate your issue on my computer. My understanding was that the auto-reconnect feature shall be activated when the microcontroller is physically disconnected from the computer and connected again. Here is a video of the auto-reconnect feature doing its thing on my computer:

https://user-images.githubusercontent.com/4225542/138178957-b0ff9048-b1a8-47cd-97fa-e0f547408d47.mp4

Is this the desired behavior of the auto-reconnect feature?

closed 2022-02-07 by alex-spataru