Serial-Studio/Serial-Studio #74: UDP example

marcobrianza, 2021-10-06:
I'm sending raw UDP packets to Serial Studio but they are not received.

It is not clear to me what is the "Host" filed meaning for the UDP connection, Is it the address where serialStudio binds locally?

can you please provide and example?

thank you
Marco

alex-spataru, 2021-10-08:
Hi,

I implemented support for both socket types (TCP/UDP) because Qt provides an abstraction layer that allows me to threat sockets as files (e.g. use `read()` and `write()` functions). However, I did not take into account UDP sockets have specialized functions for connecting/binding to a host, which seems to cause a lot of issues with the UDP socket.

I just made some changes & tests that allow Serial Studio to communicate with UDP multicast groups, and will perform tests on P2P connections tomorrow.

I'll keep you updated on my progress on this thread.

alex-spataru, 2021-10-08:
@bigjohnson Btw, thanks a lot for your sketch! If I get an `esp8266` I will totally test it!

alex-spataru, 2021-10-20:
I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.0) that fixes this issue. Please let me know how it works for you!

alex-spataru, 2021-10-21:
Hi @marcobrianza,

If I understand correctly, you would like to have an UDP configuration pane similar to this one:

[image]

Can you please confirm if this is the desired behaviour?