Serial-Studio/Serial-Studio #82: Open CSV locking up the program

Brad-HES, 2021-11-19:
I noticed when trying to view some of my CSV files that Serial Studio would freeze. It would lock up right after you selected the file to open. 
  By running Serial Studio from a command prompt, I was able to get this error message right when it locked up.

QFont::setPixelSize: Pixel size <= 0 (-1)

While trying to duplicate it, I noticed the problem appeared when I deleted a field from the JSON Project, leaving a frame index undefined. When I went back into the JSON, and made sure every single frame index within the data frame was defined, the problem disappeared. 
  Hopefully this is enough info for you to work with. 

  Thanks for making a very useful tool.

alex-spataru, 2021-12-07:
Hi! I did some changes in the CSV module to address your issue. Instead of relying on the JSON file to generate and open the CSV, Serial Studio shall now directly write RX data from the serial port to the CSV file (obviously, ensuring that each field/dataset is separated with a comma character).

This approach reduces code complexity and can also reduce CSV file size for complex JSON projects. Moreover, if you change the JSON file, you won't run into strange issues or crashes while opening the CSV.

The only disadvantage to this new approach is that newer versions of Serial Studio will have difficulty interpreting CSV files from previous versions of Serial Studio, especially if the dataset indexes in the JSON map are not defined in the same order as the frame fields.

I'll let you know when I draft a new release containing this fix.

alex-spataru, 2021-12-07:
Hi, I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) that should fix this issue (as described in the previous message). Please let me know how it works for you!

Brad-HES, 2021-12-07:
That was quick, thanks! This has fixed that problem. The program no longer locks up when trying to load the CSV. 

It is very picky when loading the CSV though. I have had to edit some of the CSV to make them work. The issue is basically the same as what you are describing, with it not being able to look at the older CSV files. 

My data stream has some lines of header info with firmware, serial number, startup system check info. This does get put into the CSV file, which is good. However, if those lines of data are in the CSV file when it tries to read the CSV, it doesn't open. The program doesn't lock up though, which is good.
    But if I just stream direct from the serial port, it seems to work fine. It receives those lines of data, and appears to ignore them. Once it gets to the data that matches the JSON project, it begins displaying data like it should.  I can delete the rows of data from the CSV and get it to work.  Is there a way to get the CSV reading to just ignore and move to the next frame, like the serial receive does?

alex-spataru, 2021-12-08:
Hi! If possible, please share an example CSV file and your JSON map so that I can work on the issue. In theory the CSV player ignores the first row of data (which corresponds to the column titles). The only validations that Serial Studio does before parsing the CSV files are:
- Verify that the first cell (`A1`) is equal to `RX Date/Time`.
- Check that all rows have the same length as the first row.

Brad-HES, 2021-12-08:
Thanks for the explanation. I think that explains was preventing the proper loading of the CSV. It wasn't directly tied to the headers, although removing them made it work. It looks like it is failing the check where Serial Studio checks that all rows have the same number of items as the first row. 

I could go back into the Serial Studio output CSV file using excel, open and resave it, and the copy would load fine. I compared the 2 files and found that when excel saved it, it added in the extra commas to fill each line to the proper number of fields. The CSV file created by Serial Studio didn't have the extra commas in it to fill out the rows to get the proper number of fields. 

This is the file as saved from Serial Studio. It would not open in SerialStudio:
RX Date/Time,Field 1
2021/12/08/ 09:32:04::978,
2021/12/08/ 09:32:05::245,
2021/12/08/ 09:32:05::284,0200; rev:       B; SN:    0022
2021/12/08/ 09:32:05::404,100200 PCB BENCHTEST_v1.0.6
2021/12/08/ 09:32:05::618,
2021/12/08/ 09:32:05::645,DB1:E103
2021/12/08/ 09:32:06::630,
2021/12/08/ 09:32:06::751,.8,00.16,000.1,000.00,00.7,00.0,30.3,159.5,159.4,159.5,159.4,00.0,021.5,0.00,000.0,26.0,1
2021/12/08/ 09:32:07::778,0.8,00.17,000.1,000.00,00.7,00.0,30.3,159.4,159.5,159.5,159.4,00.0,021.6,0.00,000.0,26.0,1

This is the file after opening it in Excel and resaving it. It opened properly in Serial Studio:
RX Date/Time,Field 1,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:04::978,,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:05::245,,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:05::284,0200; rev:       B; SN:    0022,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:05::404,100200 PCB BENCHTEST_v1.0.6,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:05::618,,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:05::645,DB1:E103,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:06::630,,,,,,,,,,,,,,,,,
2021/12/08/ 09:32:06::751,0.8,0.16,0.1,0,0.7,0,30.3,159.5,159.4,159.5,159.4,0,21.5,0,0,26,1
2021/12/08/ 09:32:07::778,0.8,0.17,0.1,0,0.7,0,30.3,159.4,159.5,159.5,159.4,0,21.6,0,0,26,1

Would it make sense to pad the data lines with enough commas so that headers or partially received data lines wouldn't cause a problem? Or alternately allow Serial Studio to just ignore incomplete lines?

alex-spataru, 2022-02-07:
Hi, I made some changes on the import/export code. If possible, please check if the [latest release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.6) fixes your issues.

closed 2022-02-18 by Brad-HES