Serial-Studio/Serial-Studio #83: JSON Object Names

davewhitters, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
Working with JSON files with Serial Studio is more difficult due to the object names being single letters in most cases.

**Describe the solution you'd like**
Descriptive JSON object names. E.g. "t" is title. 

**Describe alternatives you've considered**
Instead of using "t" for title, allowing the word "title" would make the JSON files much more digestible.

alex-spataru, 2021-11-23:
Hi! I just implemented this request with backwards compatibility with the previous JSON format. Here is an example JSON project processed with the latest commit of Serial Studio:

```json
{
    "frameEnd": "*/",
    "frameStart": "/*",
    "groups": [
        {
            "datasets": [
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 1,
                    "min": 0,
                    "title": "Runtime",
                    "units": "ms",
                    "value": "%1",
                    "widget": ""
                },
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 6000,
                    "min": 0,
                    "title": "Speed",
                    "units": "RPM",
                    "value": "%5",
                    "widget": "gauge"
                },
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": true,
                    "led": false,
                    "log": false,
                    "max": 100,
                    "min": 0,
                    "title": "Speed Reference",
                    "units": "%",
                    "value": "%3",
                    "widget": ""
                },
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 550,
                    "min": 500,
                    "title": "Duty Cycle",
                    "units": "ms",
                    "value": "%4",
                    "widget": "bar"
                }
            ],
            "title": "Data",
            "widget": ""
        },
        {
            "datasets": [
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 100,
                    "min": 0,
                    "title": "Reference",
                    "units": "",
                    "value": "%3",
                    "widget": ""
                },
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 1,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 4000,
                    "min": 0,
                    "title": "RPM",
                    "units": "",
                    "value": "%5",
                    "widget": ""
                }
            ],
            "title": "Reference vs RPM",
            "widget": "multiplot"
        }
    ],
    "separator": "",
    "title": "BLDC Test"
}
```

I'll notify you when I draft a new release containing this enhancement.

davewhitters, 2021-11-24:
This looks great, nice work!

closed 2021-11-24 by davewhitters