Serial-Studio/Serial-Studio #84: SerialStudio Segfault On First Message

tvanfossen, 2021-11-22:
**Describe the bug**
SerialStudio seg faults on receiving first message, before the dashboard has opened

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
SerialStudio to not seg fault. Have 13 tasks being collected and printed, with the last message in the stream being heap used.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Upload your log file**

Example message - I see one message print out to the log before SerialStudio crashes with a segfault

/*DurableTask 4096,0,384,2148,100,callback_task 1024,1,356,728,107,IDLE 1536,34,396,496,1176,view 25600,2,440,10564,42196,IDLE 1536,61,396,496,0,main 4608,0,404,1592,63652,RxTask 4096,0,368,708,180,Tmr Svc 2048,0,344,348,0,ipc1 1024,0,396,448,28,TxTask 4096,0,428,640,0,ProcRxTask 4096,0,436,488,0,esp_timer 4096,0,416,808,48,ipc0 1024,0,392,396,17580,Heap Used,53*/

Looping this message to send repeatedly should be enough to replicate - I counted through the set here, and everything *should* matchup to the json map

**Upload your JSON map file**
[serial_studio_debug.zip](https://github.com/Serial-Studio/Serial-Studio/files/7583684/serial_studio_debug.zip)

alex-spataru, 2021-11-25:
Hi, did you try to use software rendering? In Serial Studio 1.1.2 you can change the rendering engine in `Setup` → `Settings` → `Rendering Engine`. I am also experiencing some nasty crashes on Windows 11 when using OpenGL or Direct3D as rendering engine.

alex-spataru, 2021-11-29:
I just found out the root cause of the segfault. The problem was that creating external windows for all the displayed widgets during a short period of time put too much pressure on the rendering engine. The solution was to only generate the external window (by using an asynchronous QML loader) when the user requests a widget to be shown externally.

I´ll notify you when I draft the next release of Serial Studio to confirm if this fixes your issue too.

alex-spataru, 2021-12-05:
This issue is very closely related to #85, the rendering engine receives too much pressure when handling the "slide" animation between the console and the dashboard tabs while initializing and rendering all the widgets. Forcing widgets to render as framebuffers instead of images and adding a 500 ms delay before performing the slide seems to fix your issue and also reduces CPU/GPU usage drastically.

alex-spataru, 2021-12-07:
Hi, I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) that should fix your issue. Please let me know if it works!

tvanfossen, 2021-12-08:
Yes, the fix worked

closed 2021-12-08 by tvanfossen