Serial-Studio/Serial-Studio #85: Extremely Slow Behavior

MendesLL, 2021-12-02:
**Describe the bug**
UI will act slow and laggy upon connection, independent of render engine used. GPU usage is strangely high during setup(integrated Vega 10 running at 60-70% usage), then, upon succesfull connection, GPU usage drop to baseline and cpu usage goes all the way up(ryzen 7 3750h). May be caused by CPU incompatibility, or by Nvidia Opitmus incompatibility, or dashboard is forced to software render by some unknown factor.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a arduino leonardo to read its three adc channels and send the readings through serial with 115200 baud at about 90-100 Hz report rate, with the PC hardware mentioned
2. Set up the project within Serial Studio
3. Connect
4. Watch it as things go south

**Expected behavior**
The software should go unresponsive or just extrmely slow

**Screenshots**
None

**Upload your log file**
None

**Upload your JSON map file**
(Not necessary, I used the json editor wizard to plot 3 graphs, and 2 gauges, 1 bar/level)

alex-spataru, 2021-12-02:
Hi, does this issue occur with [Serial Studio 1.0.23](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.0.23)? I think that the root cause of the problem is the rendering method that I am using to draw Qwt-based widgets into the QML interface.

Also, if possible, please try the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) release to see if it helps with performance. I am currently experimenting (and fixing) several issues on Windows with the latest continuous release (even if it works perfectly fine on macOS & Linux). 

I strongly advise you against using the continuous release for everyday task. But having some feedback on the development version certainly helps me to improve Serial Studio & eases the process of fixing new issues.

Thanks a lot for the report!

MendesLL, 2021-12-02:
Hello, thanks for the dedication. I've seen that you really take care of this project.

Well, Since you pointed it out, I'll try 1.0.23 and continuous release, and get back with results in a few hours. I'll keep in mind the possible instability of the continuous release, but I'll use it every now and then, even if just to give some feed back. 

I'm very interested on the project, cause it's simple, versatile, and plug and play. We are developing a data aquisition hardware here at the university, and I'm trying to push Serial Studio as the visualizer/logger.

MendesLL, 2021-12-02:
The continuous release (1.1.3) behaves a little better for little time. It will trigger cpu renddering once connected (GPU usage goes 0, CPU sky high), and update GUI and respond to commands periodically, but after some time, just refuses to run or crash, the procces will just stay there, frozen, it'll use a tiny bit of RAM, and task manager struggles to kill it, need to kill the procces under the Details tab.

1.0.23 Also a bit poblematic. I by the behavior it might have a memory leak, RAM usage boomed to 1GB+ before it froze, on a second attempt, it got me black screened and forced some apps to restart, and crashed one of my browser's open tabs.

I don't know much about the subject, but maybe the parser need some kind of watchdog, and, well, either the render engine or the parser are going a bit too hard on their computations. Maybe I'm perceiving that it's rendering through CPU cause some subroutine is causing the whole application to freeze.

alex-spataru, 2021-12-05:
Hi @MendesLL, I just made some changes taking into account your observations. Here is my current CPU/GPU/RAM usage with the latest commit:

[image]

If possible, please re-download the [continuous release](https://github.com/Serial-Studio/Serial-Studio/releases/continuous) again and check if it behaves better for your setup.

MendesLL, 2021-12-05:
Now it's working a lot better, no crash whatsoever. CPU/GPU usage have gone to levels in line with what one would expect, but I am still experiening intermittent behavior, RAM will oscillate a bit everytime the UI updates. It will respond to commands (hide/show graphs and widgets, disconnect), click whenever you want, it will update within the next "time window"

alex-spataru, 2021-12-05:
> Now it's working a lot better, no crash whatsoever. CPU/GPU usage have gone to levels in line with what one would expect.

Glad to hear that! 

> but I am still experiening intermittent behavior, RAM will oscillate a bit everytime the UI updates.

I'll check if there are any memory leaks with Valgrind over the course of the week. Hopefully I can find something to fix. I'll keep you updated on this thread.

MendesLL, 2021-12-05:
> I'll check if there are any memory leaks with Valgrind over the course of the week. Hopefully I can find something to fix. I'll keep you updated on this thread.

May be the case. Since I'm using an arduino, it's possible to see when its communicating, and I observed that it would transmit, the RAM usage would go up by 80-100 MB, UI updates, Arduino stop sendind (as if TX buffer is full and PC isn't reading),

Now, since it isn't nearly real time, theres a lot of middleware here, tx start, tx stop and whatever is freezing the program mught be out of order on my observation

alex-spataru, 2021-12-07:
Hi, I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) that should improve the CPU usage and GPU usage. I will work on reducing the RAM usage for the next release.

closed 2022-02-07 by alex-spataru