Serial-Studio/Serial-Studio #86: Not receiving data via serial connection

datoslabs, 2021-12-03:
**Describe the bug**
Hi,

I'm new to Serial-Studio; I have read great reviews online about Serial Studio and am looking forward to using it for my project.  I've installed V1.1.2 on Windows 10 and after connecting to my Arduino Nano  33 BLE Sense via serial com port, it was not displaying any data on the console nor the dashboard.  I am using a JSON project file and the device is sending plain CSV formatted data (e.g. /*%s,%s,...*/). Is there a log file I can find to help debug the issues?

**To Reproduce**
Steps to reproduce the behavior:
1. Click on "Parse via JSON project file" (included below) under "Setup" window.  
2. Click on "Serial" tab and select the COM port as well as "Baud rate" for my Arduino device (in my case COM3 and 115200) 
3. Click "Connect"
4. No error message, no data output in the Console, and the Dashboard button is disabled.

I have verified that my device is outputting data via COM 3 in the Arduino IDE serial monitor, sample as follows:

`/*-0.63,-0.50,0.69,31.34,100.39,-15.69,26.02,13.00,-46.77,299.95,68.49,117.75,4340278*//*-0.58,-0.30,0.65,93.56,73.82,-16.61,25.60,12.50,-47.66,302.88,68.19,121.93,4355871*//*-0.54,-0.20,0.71,113.51,52.17,-39.08,26.18,12.61,-48.76,305.43,68.30,126.01,4370970*//*-0.55,-0.26,0.78,91.83,62.70,-70.83,26.33,11.37,-48.43,309.32,68.61,131.06,4386633*//*-0.60,-0.30,0.78,65.45,78.77,-94.02,25.62,11.84,-47.45,314.16,68.86,136.52,4401229*//*-0.54,-0.38,0.78,34.32,86.84,-94.43,26.46,12.93,-47.78,319.70,68.86,142.23,4416952*//*-0.33,-0.49,0.70,-41.59,108.94,-70.28,27.31,12.82,-45.94,324.43,68.23,146.04,4431371*/`

**Expected behavior**
Ability to view data in Dashboard.

**Screenshots**
[image]

**Upload your log file**
Unable to locate the "bug" icon on the right side of toolbar (see screenshot).

**Upload your JSON map file**
[Arduino_Serial_IMUs_Serial_Studio.zip](https://github.com/Serial-Studio/Serial-Studio/files/7650600/Arduino_Serial_IMUs_Serial_Studio.zip)

datoslabs, 2021-12-03:
Update: it seems to be working now after I re-uploaded the sketch and restarted Serial-Studio.  The output remained the same, I am not sure why it didn't work before.  Would very much appreciate it if you could point me to where I could find the log file for future references.  Thank you!

alex-spataru, 2021-12-05:
> Update1: It seems to be working except it's not parsing/displaying field indexes 10-13 after I re-uploaded the sketch and restarted Serial-Studio. The output remained the same, I am not sure why it didn't work at all before and why fields 10-12 aren't parsed at all and field 13 does not seem to contain the right value.

Hi! This is a known issue (#79), it has been fixed in Serial Studio 1.1.3 (which is the upcoming release, I just need to confirm that it fixes issues #84 and #85 and that the MQTT module works with secure/encrypted connections #34). 

If you want, you can try the [continous](https://github.com/serial-studio/serial-studio/releases/continuous) release to see if it fixes your issues.

alex-spataru, 2021-12-07:
Hi, I just created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) that should fix your issue. Please let me know if it works!

closed 2021-12-07 by datoslabs