Serial-Studio/Serial-Studio #87: Send data back to Arduino

thalesmaoa, 2022-01-01:
**Is your feature request related to a problem? Please describe.**
Sure. The first problem you clearly solved, which is read data from the uC. However, sometimes the dashboard also needs to control the uC. Let me use an example. I have to control a servo to throttle an ICE. I'm having a lot of trouble just to produce a dash to control it. A gauge from 0 to 180deg would solve it.

**Describe the solution you'd like**
Just add widgets to be controlled by the user and send data back to the uC.

**Describe alternatives you've considered**
I have an old solution like your CanSat Software using flask, html5 and python. However, it gives me a lot of trouble just to adjust it. Just wanna something simple and straight like your Serial-Studio. Alternatives are NodeRed and LabView. I don't feel comfortable of using those bazucas to kill my mosquitto.

**Additional context**
The first google image that I found:
[image]

alex-spataru, 2022-01-02:
Hi, I will begin working on this issue for the next release cycle. Right now I am currently fixing some memory-related issues and working on issue #82.

However, we can begin discussing the `Serial Studio` → `Microcontroller` frame format. For the moment, I am thinking about adding another combobox with available controls in the JSON Editor, here is a quick (and ugly) sketch:

[image]

Of course, users will also have the option to select a `None` option, which will indicate Serial Studio that it should not create any controls for the selected dataset.

Afterwards, control values could be sent through the serial port with the following format:

`/*[value for dataset 1], [value for dataset 2], ..., [value for dataset n]*/`

I imagine that the frame should be transmitted in one of following ways:
- User selects a transmission frequency/interval from the JSON editor.
- Frame is automatically sent when one of the controls is changed by the user.

Please let me know if you have any ideas or feedback regarding this implementation idea! Happy new year btw!

thalesmaoa, 2022-01-03:
Awesome, thanks for your quick reply.
I will think more about it, but looking at your suggestion seems adequate.

RasmusB, 2022-02-14:
Hello,

Thanks for the great work on your software! I'm interested in this feature as well. Is there a test build with this feature to help out with testing?

alex-spataru, 2024-11-17:
Please take a look at the new [actions](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#action-view-when-an-action-is-selected) available in the Project Editor. While for the moment this feature only enables click-and-send commands, in the future maybe we can add more widgets (such as sliders).

closed 2024-11-17 by alex-spataru