Serial-Studio/Serial-Studio #88: Ability to Process (Combine, Bit Manipulation, Arithmatic) Received Data

alperyazar, 2022-01-11:
**Is your feature request related to a problem? Please describe.**
For very low end platforms, it may be difficult to generate ready-to-display data to Serial Studio. For example, it may be hard to do floating arithmatic on the platform. Instead, the platform could send raw data. For example, suppose that we are reading a temperature sensor and the real temperature value is calculated by Temp=1.5*x + 6 where x is the raw value read from the sensor. 
The platform may send only value of x without doing the arithmatic and conversion could be done by Serial Studio, on PC, without any computation problem.

I was using version 1.0.20. Although it wasn't documented I was using a `.json` file (manual mode) like that:

```
...
"v": 256*(%1)+(%2)
...
```

to combine two fields to create a single one

or

```
...
"v": (%1)*(3.3/192)
...
```

to convert raw value to a meaningful value (in this case voltage).

I was happy with 1.0.20 but I noticed that it crashes when number of fileds send by the platform exceeds 100. Then I decided to give a try the latest version. It looks like the latest version doesn't have a problem with number of fileds but previously working conversions stopped working (I converted to the new JSON format).

**Describe the solution you'd like**
A concept like *virtual fields* may be introduced. Instead of directly mapping between dashboard `value` field with received data a layer which processes fields can be solution. Outputs of this layer are called *virtual fields*. For example, multiple fields can be combined to generate a virtual value, or can be preprocessed. Being able to do bit manipulation on received values (bit masking or logical operations) will be also good.

**Describe alternatives you've considered**
An alternative could be bringing (unintended ?) feature in 1.0.20 back or adding new JSON key to dataset element like `operation` but I prefer defining a concept like *virtual field*.

**Additional context**
Virtual fields may be prefixed by `v` and with additional JSON compatible definitions, they can be defined. For example,

```
...
"virtualfieldno": 1
"expression": "(%1)+(%2)
...
...
"datasets":
...
"value": %v1
```

alperyazar, 2022-01-11:
I was also using fields in title generation in 1.0.20 like `"t": "Title (%1)` which doesn't work with 1.1.4. Fields other than `value` may be become compatible with suggested *virtual field* concept.

alex-spataru, 2022-01-24:
Hi, I just made some commits today to add again support for embedding JavaScript expressions in the `value` field. I will notify you when I draft a new release that contains this quick fix.

Previously, I removed support for evaluating the `value` field as one of the possible fixes for issues #85 and #84. However, I forgot to re-enable the feature after finding the root cause of the crashes & slow behavior.

The high CPU/GPU usage was caused by inefficient management of the render engine. And the RAM usage has been reduced after catching some memory leaks with valgrind and enabling the [address sanitizer](https://en.wikipedia.org/wiki/AddressSanitizer) for debug builds (these fixes where made after the release Serial Studio 1.1.4). 

I'll take a look at implementing the virtual fields concept after creating a new release (to avoid pushing to many commits/features/fixes at once without testing & feedback from the community). 

I think that the JS engine included with Qt should support bitwise operations, I'll make some tests during the course of the week and provide updates/notes in this thread.

alex-spataru, 2022-02-07:
> Hi, I just made some commits today to add again support for embedding JavaScript expressions in the value field. I will notify you when I draft a new release that contains this quick fix.

I just drafted a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.6) that brings back the JS-expression evaluator for the `value` field. I'll take a look at implementing the virtual fields concept for the next release.

alperyazar, 2022-02-07:
Thank you for your interest and the new release! I will check it with my current setup ASAP and give feedback.

I am also happy you considering *virtual fields* concept.

alperyazar, 2022-02-15:
I downloaded v1.1.7 today. I noticed that CPU usage is significantly lower than my previous version (1.1.4 ?) and GUI is smoother.

Regarding this issue, previously I set `title` field as `"title": "%1"` and this didn't work with the latest version. I have another problem and that problem may be triggering another problems. With my current setup, I use manual JSON mode. Currently, there are 113 fields in the message. I suspect that Serial Studio begins to behave **unstable** when number of fields exceed 100. Is there a digit limit like 2, limiting maximum field index to %99? In my case, most of fields are shown in wrong locations but some fields are OK. Since fields are not shown in correct places, I can't confirm the original problem yet. Do you have any idea about this? BTW, I use TCP socket mode and with Wireshark I can confirm that whole data is in a single TCP packet (no fragmentation). I am using default delimiters `/*`, `*/`. Do I need LF, CR, CRLF or NONE at the and? I tried 4 of them but nothing changed.

Thank you.

alex-spataru, 2024-11-17:
**Note:** All of these features can now be done through the [frame parser function](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view), which allows users to do all the calculations they deem necessary for their projects. See also the [Hexadecimal ADC](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/HexadecimalADC) for an example on how to process binary data on Serial Studio. Frame index is now stored in a separate field in the dataset structure, which simplifies a lot the frame/dataset mapping code.

closed 2024-11-17 by alex-spataru