Serial-Studio/Serial-Studio #89: Save csv have no title

wulinzhou, 2022-01-12:
CSV files show Field 1, Field 2,Field 3 and so on

Below is my json files
{
    "t":"wireless",
    "g":[
       {
          "t":"pmic",
          "d":[
             {
                "t":"vbus",
                "v":"%1",
                "u":"mV"
             },
             {
                "t":"vsys",
                "v":"%2",
                "u":"mV"
             },
             {
                "t":"vbat",
                "v":"%3",
                "g":true,
                "u":"mV"
             },
             {
                "t":"ibat",
                "v":"%4",
                "g":true,
                "u":"mA"
             },
             {
                "t":"state",
                "v":"%5"
             },
             {
                "t":"temp",
                "v":"%6"
             }
          ]
       }
 ]
}

alex-spataru, 2022-01-21:
Hi, this was a temporary fix for issue #82, I have added support for showing dataset titles with the latest commit. I'll notify you when I draft a new release.

alex-spataru, 2022-02-07:
Hi, I just drafted a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.6) that writes dataset titles for the exported CSV file. Please let me know if I should close this issue.

closed 2024-11-17 by alex-spataru