Serial-Studio/Serial-Studio #92: Feature request: access RX time in widgets

puetsch, 2022-01-22:
I can access the data from the serial input as %1, %2 and so on, but I haven't found a way to access the RX timestamp that is written to the CSV.

I'd love to create a plot that has the RX time/date as the x axis.

alex-spataru, 2022-02-11:
Thanks a lot for the idea, I will try to get it implemented for the next release!

alex-spataru, 2024-11-29:
It took some time, but with the latest [continuous build](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous), users can now select which dataset to use as the source for the X-axis in plots!

This opens up exciting possibilities. For instance, the microcontroller can send data like packet numbers or elapsed time, allowing you to plot values against these parameters for more meaningful insights.

To achieve this, follow these steps in the project editor:
1. Define a “Time” dataset.
2. Configure the X-axis for subsequent datasets to reference the “Time” dataset.

Here’s an example configuration in the editor:

[image]

When running, the dashboard will look something like this:

[image]

For more details, check out [issue #218](https://github.com/Serial-Studio/Serial-Studio/issues/218). Another use of this feature is creating custom plots (e.g. a [Lorenz Attractor](https://marksmath.org/visualization/LorenzExperiment/)):

[image]

Thank you for your patience,
Alex