Serial-Studio/Serial-Studio #93: More flexible data format

n-e-g, 2022-01-23:
It would be super useful to be able to process data frames containing different data.
I found this issue when trying to use serial studio in my projects that send different data in different frequencies.
E.g. packet data is tagged like `SPD=10,ACCX=3,ACCY=4,ACCZ=0` that is sent very frequently, while e.g. power status like `PWR=30` is send occasionally. In general Some packets contain data that change frequently, some infrequently. 
As currently it is required to send all data within single frame, it is kind of impractical and not that flexible.
I would like to extract particular fields and values, map them to datasets and update only these datasets.

Additionally, this would allow serial studio to be used to extract data from existing protocols without changing of message format.

I can imagine implementing this as being able to add variables like `%{SPD}` instead of `%1`, use values as replecements and remove groups that contain unfilled/missing data. Then append just that datasets.

Another possibility is to have intermediate plugin (e.g. with json file) that is able to map between tag names and indices, leaving unmatched %X field as zeros, or filter them out (probably better).

n-e-g, 2022-01-25:
After thinking about this problem, I think that the most flexible implementation would be to have intermediate component that receives frame, analyzes its content, and produces data structure that is later on filled into the data sets.
This would allow even further utilization of existing protocols as data sources. E.g. by different header byte the data format could be different, even without tagged data, like `SPD,10`, `ACC,1,2,3` and so on ...
I can imagine implementation that splits definition of data sets, layouts, etc and data format definition (2 jsons) configured independently.

Btw, I kave no qt knowledge, but I am software engineer, so I could help implementing this feature if you find it useful.

alex-spataru, 2023-10-14:
I was thinking about modifying the JSON project format to something similar to this (example):

```json
{
  "mqtt": {
    "publish": true,
    "brokerAddress": "mqtt://broker.example.com",
    "port": 1883,
    "username": "mqtt_user",
    "password": "mqtt_password",
    "publishTopic": "sensors/data",
    "subscribeTopic": "sensors/commands"
  },
  "frameTypes": [
    {
      "id": "A",
      "type": "hex",
      "startDelimiter": "0xAA",
      "endDelimiter": "0xFF",
      "readings": [
        {
          "name": "temperature",
          "dataType": "float",
          "startIndex": 1,
          "length": 4
        },
        {
          "name": "humidity",
          "dataType": "float",
          "startIndex": 5,
          "length": 4
        },
        {
          "name": "light",
          "dataType": "int",
          "startIndex": 9,
          "length": 2
        },
        {
          "name": "status",
          "dataType": "bit",
          "startIndex": 11,
          "bitIndex": 0
        },
        {
          "name": "battery",
          "dataType": "byte",
          "startIndex": 12
        }
      ]
    },
    {
      "id": "B",
      "type": "csv",
      "startDelimiter": "<",
      "endDelimiter": ">",
      "delimiter": ",",
      "readings": [
        {
          "name": "pressure",
          "dataType": "float",
          "column": 0
        },
        {
          "name": "altitude",
          "dataType": "float",
          "column": 1
        }
      ]
    },
[... 202 lines cut ...]
  Serial.print(longitude);
  Serial.print('}');
  Serial.print('\n');
}

void setup() {
  Serial.begin(9600);
  randomSeed(analogRead(0));
}

void loop() {
  sendHexFrame();
  delay(1000);

  sendCsvFrame();
  delay(1000);

  sendKeyValueFrame();
  delay(1000);
}
```

Using this new format would also solve issues #134 and #28, also, maybe we could also modify this format to address issues #104, #25 and #87.

Of course, implementing this format would mean that Serial Studio would not support the previous JSON formats, and many things would need to be changed across the code base (which I plan to do, the project grew too fast and a little revamping from the ground is beginning to be necessary). What do you think about this new format? Do you have any suggestions before I start working on this new format?

alex-spataru, 2024-11-17:
Please take a look at the [frame parser function](https://github.com/Serial-Studio/Serial-Studio/wiki/Project-Editor#frame-parser-function-view). It enables users to write custom JavaScript code for their projects, providing far more flexibility than any JSON-based protocol or structure could offer.

The core idea in this case, is to create a “global” dashboard encompassing all frame types, with the JavaScript parser updating a global array based on the detected frame type. This approach also supports building dashboards from binary data, as demonstrated in the [HexadecimalADC example](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/HexadecimalADC) and the [BLE iPhone Battery Level demo](https://github.com/Serial-Studio/Serial-Studio/issues/220#issuecomment-2474466995).

closed 2024-11-17 by alex-spataru