Serial-Studio/Serial-Studio #94: JSON legacy format loading issue

pompompom, 2022-02-08:
**Describe the bug**
I am trying to load a json map file in the legacy format ( as defined on wiki) but the dashboard doesn't load. No errors appears

**To Reproduce**
Steps to reproduce the behavior:
1. Select attached project file (renamed as json)
2. Open serial device that produces frames

**Expected behavior**
As stated in issue #83 the map file should be backward compatible

**Upload your JSON map file**
[testmodel.txt](https://github.com/Serial-Studio/Serial-Studio/files/8024149/testmodel.txt)

alex-spataru, 2022-02-10:
Hi, backward compatibility was removed in [version 1.1.4](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v1.1.4) to improve performance (parsing for both old and new fields used a lot of CPU in some configurations). 

I just converted your JSON file to the new format: [testmodel.txt](https://github.com/Serial-Studio/Serial-Studio/files/8037007/testmodel.txt)

Please let me know if it works for you!

alex-spataru, 2022-02-10:
Btw, adding an option in the JSON editor to automatically convert old-JSON files to new ones would be very useful!