Serial-Studio/Serial-Studio #95: Crash on win.10 64bits pro

anthoine001, 2022-02-08:
**Describe the bug**
using the com port at 9600b.
The dashboard is running few seconds before crash. The arduino which delivers the data is still in action.
The csv file is always 33ko (684 lines the last one is not completely written)

[image]

anthoine001, 2022-02-08:
Another point : the program don't crash when I am in the console.

anthoine001, 2022-02-09:
The crash is linked with the number of packet received by second.
Adding a 100ms delay seems to solve the issue. Question is what is the maximum flow rate ?

alex-spataru, 2022-02-09:
Hi! 
Can you please upload a minimal Arduino program and your JSON file to help me replicate your issue? 

> Another point : the program don't crash when I am in the console.

Thanks for the heads up, this eliminates the possibility of the crash being caused by the IO module, probably the crash is caused by the CSV generation module.

> Question is what is the maximum flow rate ?

Well, it depends a lot on the complexity of your JSON file and the hardware resources available for your computer. I typically test SS at 115200b with a delay of 5-10 milliseconds for all my projects.

anthoine001, 2022-02-10:
Please find the sketch spoutnik3.ino --> it runs well. It generated 20-42-26.csv with spoutnik3.json

Now if yiu tak spoutnik4.ino (same as 3 but delay is only 10ms) --> it crashed but generated 20-46-21.csv.
[20-46-21.csv](https://github.com/Serial-Studio/Serial-Studio/files/
[20-42-26.csv](https://github.com/Serial-Studio/Serial-Studio/files/8044786/20-42-26.csv)
8044785/20-46-21.csv)

Note that for every crash I had (I experimented several crash with other ino sketch on MAC OS, Win 7 & WIN 10) the csv file is always 33ko.

I hope it will help to debug.

alex-spataru, 2022-02-11:
> I send a mail but .ino and .json seem not to be allowed

I received your message and just the CSV file was uploaded. If possible, please copy-paste your code and JSON file or send an email to my [personal account (alex_spataru@outlook.com)](mailto:alex_spataru@outlook.com).

closed 2024-11-17 by alex-spataru