Serial-Studio/Serial-Studio #98: Strange Serial Port behavior with Arduino Nano Every

Kerbolosh, 2022-02-16:
## Describe the bug

If I connect my Arduino Nano Every for the first time to the PC via USB, the Serial Output in Serial-Studio is shown as unreadable gibberish. If I use another program ( CoolTermWin inmy case) to establish the Serial connection, it works correctly and subsequently, it also works in Serial Studio.

## Minimal Example on Arduino Nano Every

```cpp
void setup() {
    Serial.begin(115200);
    while (!Serial);
}

void loop() {
    Serial.println("/*1,2,3*/");
    delay(50);
}
```

Kerbolosh, 2022-02-17:
> Hi! Thanks a lot for the bug report and donation! I will get an Arduino
> Nano & proceed to replicating your issue during the course of the weekend.
> I have a couple of questions:
>
>    - Which operating system are you using?
>    - Is your Arduino Nano an original or a clone?
>
> Maybe this issue is caused by the CH340
> <https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all>
> driver needed to use clones. I recall having some issues with Arduino Nano
> clones a couple of years ago.

Hi,
It is not an Arduino Nano, but an Arduino Nano Every
<https://store.arduino.cc/products/arduino-nano-every?gclid=CjwKCAiAgbiQBhAHEiwAuQ6BkmnUXW3Dg7_rD90I8RYASXvN_P3FEpqmfAzYiMIwceAsCCkFdpD2-BoCKaMQAvD_BwE>
.
And it is from the official store, not a clone. I am not sure which Serial
chip is one this Arduino, but I am familiar with some problems caused by
the CH340.
I also use the regular Arduino Nano, but I did not observe this
behavoir with the regular Arduino Nano.

The PC on which I observedit, runs on Windows 10.

Best regards
Dimitri

gregorius-ov, 2024-03-05:
I have the same issue. The issue seems to be both ways. My serial commands are not understood by the arduino as well. I first have to start the serial monitor in the IDE, then close it and start my other serial monitor for it to work.
I am using HTerm for serial communication.
Weirdly enough it seems to work when using a python script I've written which directly connects to the port.
Connecting the Arduino to the USB and using my HTerm leads to gibberish. If I start the Arduino IDE Serial Monitor first, it works.

alex-spataru, 2024-08-05:
Thanks for @recursivenomad for adding a possible fix for this issue. The latest build adds a checkbox that allows users to send the DTR (Data Terminal Ready) signal upon connecting to a serial port device:

[image]

Please let me know if the latest [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build fixes this issue!

closed 2024-11-17 by alex-spataru