Serial-Studio/Serial-Studio #101: Dashboard Greyed out on Windows

beserge, 2022-04-22:
## Describe the bug

I've got serial studio upgraded to 1.1.7, with qt 5.15.2. It was installed from the download on the site.
The dashboard button is greyed out, as is the Dashboard option under the view dropdown.
I've also tried connecting a serial device. The console works fine, but I'm still unable to access the dashboard.
I'm sure this is user error, but the documentation doesn't explain any extra steps needed to use the dashboard.

### Screenshots

[image]

[image]

## Upload your JSON map file

I tried without custom json.
I also tried loading the example json from [the wiki](https://github.com/Serial-Studio/Serial-Studio/wiki/JSON-definition-file-example)

## Upload minimal Arduino program to replicate your issue

I'm using a custom MCU, but it's just printing some basic serial data, nothing fancy.

alex-spataru, 2022-04-24:
Hi! You need to create a JSON file specific to the data format that you are sending. You can easily create and manage JSON files using the built-in JSON editor:

[image]

SammiMD, 2022-07-25:
I have the exact same issue. I'm running Serial Studio v1.1.7 and Qt Design Studio v6.2.4. I am following the Minimal Example from here: https://create.arduino.cc/projecthub/alex-spataru/serial-studio-a-dashboard-software-for-serial-devices-0866b7.

I load the JSON file from my computer, select the correct COM port and Baud rate, then press Connect. Serial Studio connects, and I get the data displayed in the Console, but the Dashboard button remains greyed out and I get no visual display.

[image]

Any help with this would be fantastic, because right now I feel kind of stupid because I can't even get the example code to work  😄

zoomastigophore, 2022-08-10:
Hello,

I had the exact same issue on Windows. I fixed it by adding a println(); at the end of the Arduino code.

```
void loop() {
   // Read voltage @ ADC_PIN
   int adc_value = analogRead(ADC_PIN);
   float voltage = adc_value * (5.0 / 1023.0);

   // Send current ms & reading through serial
   Serial.print("/*");        // Frame start sequence  [/*]
   Serial.print(millis());    // Add MCU runtime       [ms]
   Serial.print(",");         // Separator character   [,]
   Serial.print(voltage);     // Add voltage           [V]
   Serial.print("*/");        // Frame finish sequence [*/]
   Serial.println();  // <- Print an end of line, fixed the issue for me.
   
   // Wait 50 ms
   delay(50);
}
```

eedv, 2022-08-15:
Same here. Tried the minimal example, didn't work. Tried an already made json from MD_SmartCar arduino library configuring delimiters and also nothing.

RubenSDev10, 2022-08-26:
I've found the problem debugging the source code, and I think that it basically consists in that the example JSON file is not updated to the latest required format.
To fix the dashboard not appearing, simply replace the example JSON file with this one (which as you can see uses other format).
I think it would be a great idea to put an example file like this one in the JSON directory, so people can get it running quickly.
```json
{
    "frameEnd": "",
    "frameStart": "",
    "groups": [
        {
            "datasets": [
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 128,
                    "graph": true,
                    "led": false,
                    "log": false,
                    "max": 0,
                    "min": 0,
                    "title": "Runtime",
                    "units": "",
                    "value": "%2",
                    "widget": ""
                },
                {
                    "alarm": 0,
                    "fft": false,
                    "fftSamples": 128,
                    "graph": true,
                    "led": false,
                    "log": false,
                    "max": 0,
                    "min": 0,
                    "title": "Packet count",
                    "units": "",
                    "value": "%3",
                    "widget": "bar"
                },
                {
                    "alarm": 4.3,
                    "fft": false,
                    "fftSamples": 128,
                    "graph": true,
                    "led": false,
                    "log": false,
                    "max": 4.3,
                    "min": 3.6,
                    "title": "Battery voltage",
                    "units": "",
                    "value": "%4",
                    "widget": ""
                }
            ],
            "title": "sdfdssf",
            "widget": ""
        },
        {
            "datasets": [
                {
                    "alarm": 80,
                    "fft": false,
                    "fftSamples": 128,
                    "graph": false,
                    "led": false,
                    "log": false,
                    "max": 80,
                    "min": 0,
                    "title": "Temperature",
                    "units": "",
                    "value": "%5",
[... 232 lines cut ...]
            "widget": ""
        }
    ],
    "separator": ",",
    "title": "%1"
}
```

To be clear, the corresponding Serial string and Arduino code would be:
```cpp

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("/*KAANSATQRO,2051,2,5,26,10,101.26,27,32,1001,21.1619,86.8515,10,4,1.23,9.81,0.23,0,0,0*/");
  Serial.println();
  delay(50);
}
```

Also, a file like this can be generated or modified in the included JSON editor.

SammiMD, 2022-08-31:
The updated JSON file fixed my issue. Thank you so much!

savyasachi, 2022-11-10:
I have the same issue of "Dashboard" being disabled (greyed out). My json is the following

{
  "t": "My Data",
  "g": [
    {
      "t": "ADC data",
      "d": [
        {
          "t": "CH1",
          "v": "%0",
          "u": "V"
        },
        {
          "t": "CH2",
          "v": "%1",
          "g": true,
          "u": "V"
        }
      ]
    }
  ]
}

The scrrenshot of Serial Studio is
[image]

I do understand where is the issue. Why is this simple case not working.

alex-spataru, 2024-11-17:
Updated the [wiki](https://github.com/Serial-Studio/Serial-Studio/wiki/) and added several code [examples](https://github.com/Serial-Studio/Serial-Studio/tree/master/examples/) for new users. Hope you find these resources helpful.

closed 2024-11-17 by alex-spataru