Serial-Studio/Serial-Studio #102: CSV Export not working

marseeel, 2022-04-25:
Hey, I have a problem trying to export the csv. file (on a mac). When having 'create csv file' turned off, the serial output works as expected (a simple plotter), but as soon as I try to do it with it enabled, all I get is this message:

[image]

I then have to force quit the application, because the error message just pops up again after pressing 'Ok' or the 'x'. 
Do I do anything wrong, or could this be a problem with macOS?

### The JSON. File
{
    "frameEnd": "",
    "frameStart": "",
    "groups": [
        {
            "datasets": [
                {
                    "alarm": 4000,
                    "fft": false,
                    "fftSamples": 1024,
                    "graph": true,
                    "led": false,
                    "log": false,
                    "max": 5000,
                    "min": 0,
                    "title": "Oszilloskop",
                    "units": "V",
                    "value": "%1",
                    "widget": ""
                }
            ],
            "title": "Graph",
            "widget": ""
        }
    ],
    "separator": "",
    "title": "Oszilloskop"
}

### The arduino File
void setup() {
  Serial.begin(1000000);
}

void loop() {
  int val = analogRead(1);
  Serial.print("/*");
  Serial.print(val);
  Serial.println("*/");
  }

// note: the automatic formating removes the '*' after and before the '/'. (I used the default beginning/end statement)

alex-spataru, 2022-05-01:
Hi! Did you check the permissions of the folder in which you are storing your CSV files? It is probable that you (or another program) modified the read-write permissions of the `Serial Studio` folder.

You should be able to fix this by inputting the following commands in a Terminal window:

```
cd Documents
find "Serial Studio" -type d -exec chmod 755 {} \;
find "Serial Studio" -type f -exec chmod 644 {} \;
```

If you right-click the `Documents/Serial Studio` folder, the permissions should appear like this:

[image]

Please let me know if this fixes your issue!

marseeel, 2022-05-04:
Yes it did! I had shared my document folder with file sharing, and that caused the problem. Thank you very much, it now works really well!

closed 2022-05-04 by marseeel