Serial-Studio/Serial-Studio #105: Number of fields limit?

thestumbler, 2022-05-08:
I’ve resumed some testing with my UDP telemetry application. My first attempt kept crashing — it has 57 fields in one big group. I did a test and shortened it to about 10 fields, and no problem.

What is the limit? Can I get more fields by distributions them in various groups? That would actually make sense. 

Can a field appear in multiple groups?

alex-spataru, 2022-05-08:
Hi Chris,

> What is the limit? Can I get more fields by distributions them in various groups? That would actually make sense.

In theory there should be no limit in the number of fields, but I haven't done tests with very large frames. If possible, please share your JSON map file so that I can track the cause of the crash.

> Can a field appear in multiple groups?

Of course, you can even have duplicate fields in the same group (e.g. you want to display the data in multiple widgets at the same time).

thestumbler, 2022-05-08:
Okay. I’ll send it shortly. Just woke up here. I may need to obscure the field names, I’m not sure if my client thinks these are sensitive information or not. But that should be easy. 

I want to group variables anyway, because if makes logical sense. There’s s lot of stuff going on in this machine. 

What’s the json format for a multi plot? Or do you have an example?

Brief rundown on things. 

Telemetry is generated by microcontroller C++ and send over UDP. On the PC I generate a the json file using a short python script I wrote — the telemetry description already exists in Python because our PC app runs in Python, and it was easy (lazy) for me to quickly make a json file because I already was using json to save / pickle machine register settings. 

Which version ? When I look back at our UDP experiments in Feb, I’m not clear whether I was using 1.17 or continuous build when I reported “now it works”. And it’s confusing because the splash screen of both programs says 1.16 8-)

Before when I first discovered Serial Studio, it seemed that I had to put every field in the frame into the json file or it wouldn’t work. But that isn’t true now? It would seem so. 

I just thought of something. Would having an extra comma at the end of the string be an issue? 

1,2,-7,3.1415927,0,

-Chris

thestumbler, 2022-05-08:
There seems also to be a limit (or sorts) to the name of the field. Some of my names are too long, and in the overall list of values you can’t see the whole name. Even if you expand the window, it just makes the letters bigger without revealing the hidden characters. 

Example:

status_qty_apples_green_large_clean
status_qty_apples_green_large_with_worms

Displays as:

status_qty_apples_gre ….  13
status_qty_apples_gre ….  74

thestumbler, 2022-05-09:
I found one minor problem with my json file, although that wasn’t the thing that keeps it from working. To specify no frame separators, I had simply `””` in the value. 

When I looked at a json file generated by the editor, the value was `\”\”`. 

The first case is an empty string (probably indicating the default) whereas the correct one is a string consisting of two double quotes.

thestumbler, 2022-05-10:
What about the speed of the UDP packets? Mine arrive every 100 msec. I thought about this suddenly when I found a counter timer bug in my system this morning, where I was sending them every 80ms instead of 100ms (now corrected).

closed 2024-11-17 by alex-spataru