Serial-Studio/Serial-Studio #112: could show data when move cursor in plot?

hehao9051, 2022-07-09:
Serial Studio is a great tool，but i encountered some problems when use it.
I want to use this tool to show temperature and measure time of temperature change（eg: 30℃-90℃ spend 10S），this need 3 feature:
1、show data when move cursor in plot
2、pause the plot show
3、show  delt of two point (user select) in axis 
coule this feature be add in feature?

alex-spataru, 2025-04-20:
Hi, took a while to respond, but I just began implementing this. 
- A crosshair overlay is shown over the plots with with current X/Y values
- Ability to zoom and navigate a plot in a similar way to how you would use a CAD program.

Below a screen recording to show how it works. You can try the latest [continuous build to test the feature](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous).

https://github.com/user-attachments/assets/881bee00-6d75-46dd-b882-ddb2a8b58e56

Let me know what you think about it...I'll let you know when I implement data streaming pause/replay feature.

alex-spataru, 2025-04-21:
As of latest commit there is also support for pausing/resuming data without disconnecting from the dashboard. Shortcut has been set to `Ctrl + P`...`space`, `ctrl + space` is kind of tricky as it can interfere with operating system features. If anyone has a better shortcut idea please let me know!

[image]

closed 2025-04-21 by alex-spataru